mooltipass/minible #155: Automaticaly disable/enable bluetooth when card is removed/inserted

elendil-software, 2020-07-12:
### Missing feature

When no card is in the MiniBle, the should be disabled. And when the card is inserted it should automatically enabled bluetooth if it was previously enabled.
As enabling bluetooth take a few seconds, I would make this optional  

### Justification
There no need to have bluetooth enabled or connected to a device if it's not in use.
Maybe this could save a bit of autonomy ?

### Workarounds
Mannualy disable the bluetooth

limpkin, 2020-07-19:
Implementing this would require disabling / enabling bluetooth upon card insertion in order to still be able to use the _unlock_ feature. That is provided that the user computer can connect quickly enough to the device once the user has unlocked his card...

limpkin, 2020-09-05:
i think i'll make it a setting....

closed 2020-09-24 by limpkin

elendil-software, 2020-11-07:
I tried to use the "Disable BLE on Carde Removed" and "Disable BLE on Device Lock". Perfect !

commit f98d5c62 (2020-09-24): settings to disable bluetooth on lock & card removal, fix #155 & fix #129