mooltipass/minible #157: Third Field Typing Feature

limpkin, 2020-07-19:
### Missing feature
Being able to type the credential third field

### Justification
Nice to have: would enable for example typing the URL

### Workarounds
None

RChadwick7, 2021-06-16:
I second this. I recently came across a website that wants three fields... Your email or acct number, your password, AND your last name.

WilliamWicker, 2021-08-04:
And I third it! Being able to key in (or at least display) a secondary password/challenge question answer would be awesome! Of course, the natural end result is to be able to scroll through N additional responses (secondary password, challenge #1, challenge #2, etc.)

Maybe being able to associate a multi-line note with a set of credentials and then able to transmit text from a selected line of that note would be the 80% solution?