mooltipass/minible #161: Webauthn on Linux fails

Arffeh, 2020-07-24:
### Expected behavior
Webauthn on linux to succeed/not get stuck.

### Actual behavior
Webauthn on linux does not succeed.

### Step by step guide to reproduce the problem
Go to https://webauthn.me/ or https://webauthn.io/

Attempt to register

Firefox requests for registration: 

[image]

MP-BLE doesn't wake. 
Manually waking the device does not show any prompts to store credentials.
No logs are generated in Moolticute from the event.

### Moolticute Version - If Involved
0.44.4.testing.r0.g609a85a-1

### Firmware Version
AUX MCU version:  0.33
Main MCU version:  0.39

### Operating System
<sub>Glorious</sub>  Arch Linux
  

### Additional
libfido2 is installed.

udev rules:
```
# udev rules for allowing console user(s) and libusb access to Mooltipass Mini devices
ACTION!="add|change", GOTO="mooltipass_end"

# console user
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="16d0", ATTRS{idProduct}=="09a0", MODE="0660", SYMLINK+="mooltipass_keyboard", TAG+="uaccess", TAG+="udev-acl"
SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="4321", MODE="0660", SYMLINK+="mooltipass_keyboard", TAG+="uaccess", TAG+="udev-acl"
# libusb
SUBSYSTEM=="usb", ATTRS{idVendor}=="16d0", ATTRS{idProduct}=="09a0", MODE="0660", SYMLINK+="mooltipass_device", TAG+="uaccess"
SUBSYSTEM=="usb", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="4321", MODE="0660", SYMLINK+="mooltipass_device", TAG+="uaccess"
```

0x0ptr, 2020-07-28:
Hi Arffeh!

Thanks for your report. 

The reason it is not working properly on Linux/Firefox is due to the unfortunate fact that Firefox does not have support for FIDO2/CTAP2 (which is the only thing we support) on Linux (yet). 

Firefox only have support for U2F on Linux but Firefox goal is to have it done by 2020 (https://bugzilla.mozilla.org/show_bug.cgi?id=1619850)

You can confirm that this is the reason by using Chrome browser on Linux and see if that works fine. If it does we will close this report. If not we can investigate further why Chrome doesn't work.

Thanks again for your detailed report!

Tiki-God, 2020-07-29:
For me on using Chrome 84 on webauthn.io I can register a username and the minible prompts me to save it, but when I try and login on the site nothing appears to happen on the minible and the login times out.  Sometimes it'll try and store a credential for .dummy and then say I'm using a security key not registered with this website.

Another question is there a way to see information related to these credentials that are stored on the minible?

0x0ptr, 2020-08-08:
Able to reproduce the problem and a fix has been developed. See above

0x0ptr, 2020-08-09:
Hi Jon,

> For me on using Chrome 84 on webauthn.io I can register a username and the minible prompts me to save it, but when I try and login on the site nothing appears to happen on the minible and the login [...]

This should be resolved in the next firmware release. 

> Another question is there a way to see information related to these credentials that are stored on the minible?

Not yet but it is on our roadmap (maybe in 3 weeks...)

closed 2020-08-09 by limpkin