mooltipass/minible #164: To be investigated: screen blinking when connected to USB power bank

limpkin, 2020-07-29:
I'm guessing a loop between mini BLE detecting 5V, changing power source, then the power bank removing that 5v, etc etc

limpkin, 2020-08-01:
issue seems only reproducible on that particular power bank...

limpkin, 2020-09-04:
issue also happens when connecting the Mini BLE via USB through a USB-C multi adapter on a MacBook Air (https://www.amazon.fr/gp/product/B07PLSF3QJ)

limpkin, 2020-09-24:
Issue has been traced down to a bad design on the power supply: when the mini BLE is connected to USB, it consumes less than 100mA. As a result, the power bank believes that there's no USB device connected.... and switches off. Then it detects again that something is connected... etc etc

ThunderKing54, 2020-10-06:
@limpkin short question. Can the Mooltipass detect the repeating power on/off and "block" the power over USB for about a minute? If it is able to detect it  a message should appear that the plugged in device don't support chargeing of devices which draw less then 100mA.

limpkin, 2020-10-06:
@ThunderKing54 it can :)

ThunderKing54, 2020-10-06:
Woult that be a option? Display a message and block the charging when it starts flashing?

limpkin, 2020-10-06:
indeed!

limpkin, 2021-02-01:
to whomever finds this issue, this is what you want to buy: https://www.tindie.com/products/overz/smart-power-bank-keep-alive/?utm_source=twitter.com&utm_medium=social&utm_content=tweet&utm_campaign=product_back_in_stock_tweets

ThunderKing54, 2021-02-02:
Cool! I don't need that right now but it could help other people if they don't want to use other power banks or their laptop/macbook make these problems. Thank you Mathieu for finding a solution :)