mooltipass/minible #169: Extra Step in _unlock_

dehne, 2020-08-16:
### Expected behavior
When the minible is connected both by BLE and by USB and the situation calls for the _unlock_ credential to potentially be typed, the device should prompt first for whether or not to type the credential and then, only if the credential is actually to be typed, should it ask for which interface to use.

### Actual behavior
The device often prompts for which interface to use -- BLE or USB -- and then for whether or not to type the _unlock_ password.

### Step by step guide to reproduce the problem
1. Enable the _unlock_ feature on the minible and set it up.
2. On the minible, enable BLE and get it connected to a paired device. A moble phone, say.
3. Power the minible down. (First unplugging it from the USB if it's plugged in.)
4. Plug the minible into a USB attached to a Windows computer.
5. After the minible initializes itself, enter your PIN to unlock it.
6. The minible enables Bluetooth.
7. The minible prompts for which interface -- USB or BLE -- to use. Select USB.
8. The minible prompts whether to send _unlock_ password. Select do not send.
9. Ask yourself why it wanted to know which interface to use when you didn't want it to send anything ;-)

There seems to be a race condition here: Step 7 only happens if the BLE device manages to connect before the minible decides it should deal with the _unlock_ feature. With my phone as the BLE device, it happens about half the time.

### Moolticute Version - If Involved

### Firmware Version
AUX MCU version:  0.34
Main MCU version:  0.41

### Operating System
Mention if you are using either:
- Windows 7/10

limpkin, 2020-08-16:
very valid point... maybe i should enable bluetooth the moment the card is inserted...

limpkin, 2020-09-05:
I'll implement tomorrow BLE interface enabling upon card insertion. Out of curiosity, why don't you long press the wheel when you're asked to select an interface?

dehne, 2020-09-05:
Yes, the user could long-press the wheel to reject both interfaces and, thus, rejecting sending the _unlock_ credential to either interface. But at the point in the interaction where the device asks which interface to use, it hasn't told the user what information is being asked about. In all the other interactions I can think of, we first establish what information we're talking about and then deal with what we should do with it.

limpkin, 2020-09-06:
So I started implementing this but then realized it's a lose-lose situation, as depending on the user unlock feature preference, denying the login could bring up the password entering prompt... so in both cases the user would have to long press to go back (also generating "strangeness").
I'm therefore going to close this one and accept it may be a source of discomfort. However, i'll be implementing #172 so you don't get prompted when the computer already is unlocked. However, the situation will still arise again if connected to a computer or device not running moolticute :/

closed 2020-09-06 by limpkin