mooltipass/minible #171: Device not recognized after Windows Sleep Mode

ThunderKing54, 2020-08-27:
### Expected behavior
If you left your Mooltipass connected to your PC it should work normal when you come back the next day. 
### Actual behavior
The Mooltipass is not reconized anymore. You have to unplug it and plug it back in. 
### Step by step guide to reproduce the problem
1) Connect your Mooltipas swith USB
2) Enter your smartcard and pin
3) Set the Energie saving mode to:
Screen saving after 10 minutes
Hybernate after 20 minutes
4) Wait 24h and leave the Smartcard inside the Mooltipass
5) After 24h the Mooltipass is locked and your pc should turn on again when you type any key or use your mouse.
6) Enter your pin and visit any website
7) The Mooltipass is not recognized
### Moolticute Version - If Involved
v0.44.4-testing
### Firmware Version
AUX MCU version:  0.34
Main MCU version:  0.41

### Operating System
- Windows 10 LTSB 2016

### Other Notes
I test it again tomorrow and check if Moolticute detects the Mooltipass and if the Extension in Chrome detects a connected Mooltipass.

elendil-software, 2020-10-05:
I noticed the same issue. But it does not occurs only when the PC is in hibernation, it also occurs when I power off my PC. And it does not occurs every time. Usually the only way to recover is to power off the Mooltipass et restart it.

I also noticed a similar issue where the Mooltipass does not prompt for credentials. But in this case the issue is probably on the Moolticute side because if I restart the Moolticute daemon it works again.

Julien

limpkin, 2020-10-05:
Yup... need to figure out where this bug comes from.
Regarding the missing prompt, I've seen that moolticute doesn't get the latest status update... I still need to get to the root cause of this.

elendil-software, 2020-10-11:
@limpkin 
Could moolticute logs be useful for the second one ?

limpkin, 2020-10-11:
so both bugs have actually been fixed. the lack of prompt was fixed but not released yet: https://github.com/mooltipass/minible/commit/e27a647b86e39f9ecb7d685a33f9bb9a039c17aa

closed 2020-10-11 by limpkin