mooltipass/minible #200: MiniBLE hangs while pretending to enter credential management or update FW

MayaPosch, 2020-10-10:
### Expected behavior
I expect to be able to open the credential management on the MiniBLE device via the Moolticute software, as well as update the firmware to the latest version.

### Actual behavior
With the current Moolticute software, the MiniBLE device is recognised, but no confirmation dialogue ever pops up on the MiniBLE screen. When trying to do a firmware update (Ctrl + Shift + F3 to open the BLE tab), the 'upload' progress indicator continues forever, even after reconnecting the MiniBLE device.

Attempting to get information on the current device ID and firmware version via the software also never completes.

When using the same Moolticute software with a Mooltipass Mini, the process works as expected and credential management completes without issues.

Switching USB ports (directly on root controller, different root controller, USB hub/no hub, etc.) or USB C cables has no effect.

### Step by step guide to reproduce the problem
- Connect MiniBLE device.
- Attempt to open credential management.
- Wait forever while looking at the spinning animation.

### Moolticute Version - If Involved
Moolticute v0.44.12-testing.

### Firmware Version
AUX MCU version:  Unknown, not listed in device.
Main MCU version:  Unknown.

### Operating System
I'm using Windows 7 x64 Ultimate.

limpkin, 2020-10-10:
Hello Maya!
Wow, that's a bad one... 
Did you by any chance update moolticute without updating to one of the latest versions of our firmware?
What's your firmware versions?

MayaPosch, 2020-10-10:
The firmware version is the one I updated to just before I wrote the MiniBLE hands-on review for HaD. I can't seem to find out the version any more, though.

I am pretty sure that I updated Moolticute once or twice without using MiniBLE on this system, yes. Is there some kind of dependency between Moolticute and MiniBLE?

limpkin, 2020-10-11:
that's it! we usually try to avoid these dependencies, and I thought I had warned everyone about it (sorry...).
Could you revert to a previous moolticute version (from betas.themooltipass.com/) and apply the firmware update (numbered with white silkscreen) that I just sent you?

closed 2020-10-11 by limpkin