mooltipass/minible #204: "Enabling Bluetooth" message use Device Default Language

elendil-software, 2020-11-07:
### Expected behavior

Should use language set in the mooltipass settings

### Actual behavior

The "Enabling Bluetooth" message related to the "Disable BLE on Card Removed" option use the "Device Default Language".

### Step by step guide to reproduce the problem

1. Set the device language to English
2. Set the Device Default Language to french
3. Enable the "Disable BLE on Card Removed" option
4. Remove the card
5. Put it in again
6. The message is "Activation du Bluetooth" and not "Enabling Bluetooth"

This is the only case where the message is in french. When the card is removed or the Bluetooth is enable/disable from the menu, the messages are in the device set language.

### Moolticute Version

v0.44.17

### Firmware Version
AUX MCU version:  0.44
Main MCU version:  0.50

### Operating System
- Windows 10

limpkin, 2020-11-07:
wow, nice catch!

closed 2020-11-07 by limpkin

elendil-software, 2020-12-16:
@limpkin 
Just found an other case of the same kind. When the card is inserted the wrong side, displayed messages use the default language :  

- Carte invalide insérée
- Veuillez Enlever la carte
- Mise en veille
- Mise à Jour Firmware

limpkin, 2020-12-16:
shouldn't they be, if the mini doesn't know yet the language associated with said card? :)

elendil-software, 2020-12-16:
My bad, of course the mini can't know the language.
Note to myself : don't post issue to late in the evening :-D

commit 84eabde4 (2020-11-07): card insertion: bluetooth message in user's language, fix #204