mooltipass/minible #205: Linux issue: can't pair on Arch

limpkin, 2020-11-17:
Looking at journalctl -b

>Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/deviceinfo.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/deviceinfo/dis.c:read_pnpid_cb() Error reading PNP_ID value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_read_cb() Error reading Report value: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:info_read_cb() HID Information read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_map_read_cb() Report Map read failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:report_reference_cb() Read Report Reference descriptor failed: Request attribute has encountered an unlikely error
Nov 17 23:16:37 limpkins bluetoothd[1415]: profiles/input/hog-lib.c:proto_mode_read_cb() Protocol Mode characteristic read failed: Request attribute has encountered an unlikely error

It seems that due to these errors the pairing procedure fails on Linux's side (see warning in the picture below)

[image]

FYI, a correct pairing procedure looks like this:

[image]

limpkin, 2020-11-22:
So it doesn't seem like this issue was related to the impossibility to read characteristics. 
This issue was initially reported for a Linux OS running in a VM. When trying natively on a machine, the first pairing attempt does fail while the second goes through (somehow...).

haadr, 2020-11-23:
Did you pair via a GUI, and if so, which?

Or did you par via bluetoothctl?

limpkin, 2020-11-23:
both bluletoothctl and the GUI lead to the first pairing failing... 
I'm fairly certain this is not mini BLE related.

limpkin, 2021-06-19:
for those finding this bug, this was fixed with a recent bluez update (check the associated issues).
you'll likely have to compile bluez from scratch.

closed 2022-03-22 by limpkin