mooltipass/minible #210: Presence of previously paired device and absence of paired device prevent MiniBLE to enter sleep mode

elendil-software, 2020-12-16:
### Expected behavior

previously paired devices should not be able to bother the miniBLE

### Actual behavior

previously paired devices and absence of currently paired device prevent the miniBLE to enter sleep mode

### Step by step guide to reproduce the problem

- pair MiniBLE with two devices (or more)
- delete pairings
- the miniBle enter in a Going to Sleep --> Wake up cycle
- disable Bluetooth on all computer
- miniBLE finally enter sleep mode

Additional steps

- Enable Bluetooth on all previously paired PC
- Pair with one of the PC and connect
- Disable Bluetooth on the connected PC
- the miniBle enter in a Going to Sleep --> Wake up cycle

Note : 
- Sometimes when pairings are deleted it go to home screen (locked) but seems to be still connected in the Windows Bluetooth panel.
- Sometimes shows a Many Failed Connections
- Battery is going down fast (80% to 30% the time of checking and writing this procedure)

### Firmware Version
AUX MCU version: 0.46
Main MCU version: 0.55 

### Operating System
Mention if you are using either:
- Windows 10
- iOS 14.0

limpkin, 2020-12-16:
I'm going to leave this like that for the moment.... the thing is that when devices try to connect to the mini BLE, it wakes up the device (display off or on) regardless. Attracting user attention that something is odd isn't a bad idea IMHO

elendil-software, 2020-12-16:
I agree, maybe a message could let the user know that a removed paired device try to connect to the MiniBLE ?

ThunderKing54, 2020-12-17:
The Mooltipass should display a message. I was confused why my MiniBLE starts flashing ;) See #213

rastagraffix, 2022-01-16:
It seems that if an unpaired device tries to connect and the Mooltipass is not in discoverable mode, it should just ignore the request and go back to sleep. The current behaviour seems like a bit of a DoS vector in that it allows anyone to increase the battery drain, potentially locking out the user if local usb power is unavailable.

limpkin, 2022-01-16:
I'm actually going to close that issue: bundle v7 (https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/platform_defines.h#L238) actually disables bluetooth when too many failed connections. 
I chose to implement it this was as in the past the battery drain would worry users.   
FYI professional devices such as logitech keyboards, when removing pairing data, **actually switch MAC addresses** so this issue doesn't happen. I'm afraid we don't really have the budget to add X extra MAC addresses :/

closed 2022-01-16 by limpkin