mooltipass/minible #212: #004 bug smartcard (device wakes up from sleepmode all time)

ThunderKing54, 2020-12-17:
### Step by step guide to reproduce the problem
1) enter smartcard an pin
2) go to the bluetooth menu and select the delete devices button
3) pull out your smartcard very fast and put it in the wrong way around
4) wait until the Mooltipass picture appear and pull out the smartcard again.
5) The #004 Bug message appear.

### Firmware Version
AUX MCU version:  0.46
Main MCU version:  0.55

### Operating System
Mention if you are using either:
- iOS 14.2

limpkin, 2020-12-17:
I can't reproduce... could you send me a video? :)

limpkin, 2020-12-19:
fixed by https://github.com/mooltipass/minible/commit/e3a3a924e5ab3585f3d4692bfd6aa49d40e067f4

@ThunderKing54 : free mini BLE for you!

closed 2020-12-19 by limpkin