mooltipass/minible #216: Enter Debug Menu with Clone Card

ThunderKing54, 2020-12-29:
@limpkin I don't know if you're able to fix this or if the debug menu getting removed for productive units. If I think a little bit about this "bug". Can I flash the current image with that method? ;D I will try that :I 
### Expected behavior
The debug menu should not be accessable over the cloning menu
### Actual behavior
If you insert a second pre written card you're able to access the debug menu. 
### Step by step guide to reproduce the problem
1) go to the cloning menu
2) enter your current smartcard pin
3) enter the new smartcard pin
4) repeat the new smartcard pin
5) pull out your current smartcard
6) put the current smartcard back into the device
7) You revice the message that the inserted card is not empty. 
8) Now the message appear  "remove card"
9) Hold the scroll wheel for some seconds. 
10) Now you're inside the debug menu. 

### Firmware Version
AUX MCU version:  0.48
Main MCU version:  0.57

### Operating System
- Windows 10

ThunderKing54, 2020-12-29:
Update: yes you can flash the current image with that method 🤣

limpkin, 2020-12-29:
ahaha, closing this issue as the debug menu will indeed be removed for production units :)

closed 2020-12-29 by limpkin