mooltipass/minible #217: Webauthn Linux Browser support

WebmasterTD, 2021-01-13:
Webauthn on Linux (Ubuntu 20.04 LTS) is not working with:

- [ ] Firefox 84.0.2

- [ ] Chromium 87.0.4280

Works with:

- [x] Chrome 87.0.4280

- [x] Opera 56.0.3051

tested on https://webauthn.io

d4g, 2021-06-17:
Webauthn does work with yubikey on linux using Firefox 88.0.1.
Using the mooltipass ble, it does not. No message pops up on the mooltipass.

limpkin, 2021-06-17:
@d4g i'm guessing that's because it's going over CTAP and not CTAP2 :/

d4g, 2021-06-17:
It seems you are right. The relevant issue is:
https://github.com/mozilla/authenticator-rs/issues/33

0x0ptr, 2021-06-19:
> It seems you are right. The relevant issue is:
> [mozilla/authenticator-rs#33](https://github.com/mozilla/authenticator-rs/issues/33)

Yes, we have been tracking this for some time now....
See https://bugzilla.mozilla.org/show_bug.cgi?id=1530370 and https://bugzilla.mozilla.org/show_bug.cgi?id=1530373

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

0x0ptr, 2021-07-01:
> It seems you are right. The relevant issue is:
> [mozilla/authenticator-rs#33](https://github.com/mozilla/authenticator-rs/issues/33)
> 
> Is there a plan to bring CTAP to Mooltipass?

No, there is not. There is a space issue (large keys etc).

bigbellyburger, 2021-07-12:
When is it expected for this to be resolved? Might be worth noting the kickstarter promises this to work:

> You may have heard about WebAuthn. To simplify a bit, it makes passwords obsolete by authenticating you securely with signed cryptographic challenges. You simply need to approve a prompt on the device and – ta-da! – you're logged in. Yes, the Mooltipass Mini BLE supports it!

0x0ptr, 2021-07-13:
@bigbellyburger What specific issue with WebAuthn are you referring to? FIDO2/Webauthn currently works with MiniBLE. However, many web sites have not moved to using FIDO2 yet and still uses the old U2F standard which the MiniBLE does not support. Sites such as Microsoft works fine with MiniBLE for example while Google does not since Google still only supports U2F ATM.
The other limitation is that the MiniBLE does not have a secure enclave so it is using a self-signed certificate and some web sites does not allow authenticators with self-signed certificates. However, I don't know how widespread this is. (And some web sites are even more picky and for example ONLY supports keys from Yubico!)

There is also a limitation on browser support on some OS'es. Firefox on Linux/MacOS does not support FIDO2 for example (see https://bugzilla.mozilla.org/show_bug.cgi?id=1530370 and https://bugzilla.mozilla.org/show_bug.cgi?id=1530373). Chrome seems to work well on all platforms.

My1, 2023-02-21:
> I am using the thing with firefox and chromium on Fedora 36. Works totally fine for sites that arent broken for other reasons like attestation pickyness. Other fido key even works in firefox on microsoft signons.

you can try to check "anonymize anyway", which will strip the attestation away, equally if chrome asks click skip on the prompt.

VincentVanlaer, 2023-03-29:
With https://github.com/mozilla/authenticator-rs/pull/237 and https://github.com/mozilla/authenticator-rs/pull/230 merged, the backend firefox uses works with the minible.

VincentVanlaer, 2023-05-05:
:tada: I can confirm now that FIDO2 works with the mooltipass on Linux with Firefox Nightly! :tada:

limpkin, 2023-05-05:
well damn.... should we.... close that issue? :)

My1, 2023-05-05:
I'd say better wait till it hits stable. as most users likely dont use nightly

hijackeel, 2023-06-08:
[Firefox 114.0](https://www.mozilla.org/en-US/firefox/114.0/releasenotes) has been released with the fix. Tested and working for me on Gentoo!

oliverpool, 2023-06-12:
On Arch Linux - Firefox 114.0.1 it almost works:
- github.com : credential creation looks good (BLE confirmation), however nothing appears on the BLE when trying to use it for the first time (and github simply times out)
- stripe.com: credential creation looks good (BLE confirmation), however when using it for the first time, the BLE indicates `Cred. Already Registered` and stripe fails with `The operation failed for an unknown transient reason`
- gitlab.com: works fine :tada: (so maybe not a minible issue)
- codeberg.org (based on Forgejo): same a github (no message on BLE upon login) and error message `The operation failed for an unknown transient reason`
- webauthn.io: works fine

I have no idea how I can debug this at all. Apparently it does not use the moolticuted daemon for the communication.
If anyone knows how to debug/get a more helpful error message, I would be grateful :)

I will try to debug forgejo a bit more (I have done some contributions there) and try to find out the issue.

oliverpool, 2023-06-12:
On **forgejo**, it works fine if the miniBLE is the first security key, but fails otherwise (Firefox shows a prompt, but not the BLE). `T7eTAA3TODpNkJB77p1Xcg==` is the base64 key of the ble - it gets converted to Uint8Array before calling `navigator.credentials`

This works:
```json
{
  "publicKey": {
    "challenge": "nwTlBL_jxhNYHvaycLIdCL7DJW-67QAC343RpxaK2QY",
    "rpId": "localhost",
    "allowCredentials": [
      { "type": "public-key", "id": "T7eTAA3TODpNkJB77p1Xcg==" },
      {
        "type": "public-key",
        "id": "/dP5SnmJ4s/MMhcp4cSK4t0y4Ii+jz7nMjP2JXrAYbd7DC+I1r2EojYVGYTG/6me"
      }
    ],
    "userVerification": "discouraged"
  }
}

```

---

This fails: (firefox shows a prompt, but no confirmation appears on the BLE)
```json
{
  "publicKey": {
    "challenge": "dK9z_Mb34UQcOvSJKaheVU0slrrM43rJFmkfwntyhNU",
    "rpId": "localhost",
    "allowCredentials": [
      {
        "type": "public-key",
        "id": "/dP5SnmJ4s/MMhcp4cSK4t0y4Ii+jz7nMjP2JXrAYbd7DC+I1r2EojYVGYTG/6me"
      },
      { "type": "public-key", "id": "T7eTAA3TODpNkJB77p1Xcg==" }
    ],
    "userVerification": "discouraged"
  }
}

```

This explains why github and codeberg.org work in my previous comment (I have another key already registered there).

oliverpool, 2023-06-12:
I can reproduce this on https://gramthanos.github.io/WebDevAuthn/

*Create credentials*
1. Go to https://gramthanos.github.io/WebDevAuthn/credential-creation.html
2. Click `Generate options`
3. Click `Create credentials`
=> Everything works fine

*Get credentials*
1. Go to https://gramthanos.github.io/WebDevAuthn/credential-get.html
2. Click `Generate options`
3. Click `Get credentials`
=> Everything works fine

*Scroll back up and add an unknown credential AFTER*
1. Under `Allow Credentials*
2. Click `Add credential`
3. Put `T7eTAA3TODpNkJB77p1Xcg==` under `publickeycredentialdescriptor.id`
4. Click `Generate options`
5. Click `Get credentials`
=> Everything works fine

*Scroll back up and add an unknown credential BEFORE*

1. Under `Allow Credentials*
2. Click `Add credential`
3. Put the same value as the first one under `publickeycredentialdescriptor.id`
4. Remove the first one (so that your created public key is now second)
5. Click `Generate options`
6. Click `Get credentials`
=> Firefox shows a prompt, but the the miniBLE. After a while, it shows `The operation failed for an unknown transient reason`

oliverpool, 2023-06-12:
The procedure above works fine on Vivaldi (Chromium-based browser). So it is likely a Firefox issue.

I would be grateful if anyone could point me out a direction to debug further (I could not see anything interesting in `strace -f -e execve firefox`).

oliverpool, 2023-08-25:
Since https://github.com/mozilla/authenticator-rs/issues/33 was recently closed, I did my tests again (Firefox 116.0.3 - 64-bit):
- github.com: everything seems ok :tada: 
- stripe.com: `The operation failed for an unknown transient reason` upon adding the security key (after being prompted on the BLE)
- codeberg.org (based on Forgejo): everything seems ok :tada: 
- https://gramthanos.github.io/WebDevAuthn/: (see https://github.com/mooltipass/minible/issues/217#issuecomment-1587567586 for procedure): everything seems ok :tada:  (the "wrong order" one seems slower, but still works)

Regarding the remaining issue (stripe), when trying with vivaldi, I get:
`You security key does not have enough space for any more accounts.`
[image]

So something else might be going on...

Anyway I think this issue can be closed (and maybe another one opened with failing sites like stripe).

My1, 2023-08-25:
stripe may be coming down to the attestation, you can try with "anonymize anyway" active, but when you tried it for a second time, it already has created the credential on the MP, and it will block a new one with the same name, so you have to delete the other one first