mooltipass/minible #221: Nyancat screensaver gets offset to the left whenever a change is made with it active.

GlitchOne, 2021-02-08:
### Expected behavior
The Nyancat screensaver stays in the same place, or gets disabled if the change made is the one that switches the screensaver from Nyancat to None.

### Actual behavior
The Nyancat screensaver gets offset to the left after a brief moment where the screen seems to refresh.

### Step by step guide to reproduce the problem
Open Moolticute.
Connect the MiniBLE to USB.
Enter the pin.
Set the Screensaver to Nyancat
Wait a few seconds for it to start.
Click on the "Display Pin on Back Pressed" checkbox.
Click on the "Save to device" button.

### Moolticute Version - If Involved
v0.45.2

### Firmware Version
AUX MCU version:  0.51
Main MCU version:  0.62

### Operating System
Windows 10
[image]

GlitchOne, 2021-02-08:
Additionally, it seems that setting the screensaver to Nyancat and doing some other change also has other weird behaviors like:

- turning the brightness down.
- completely shutting off the screen.
- moving the image down about half of the screen and inverting the colors.

Removing the device from USB and reconnecting it seems to solve all of these issues and brings the MiniBLE back to a default state.

limpkin, 2021-02-08:
this was recently fixed, but will keep this issue opened until i'm satisfied with testing on my side.
Will publish a firmware update hopefully in a week or so.

limpkin, 2021-02-17:
recently fixed, closing it :)

closed 2021-02-17 by limpkin