mooltipass/minible #222: Feature request: Add shorter configurable lock delay.

GlitchOne, 2021-02-08:
### Missing feature

Having a shorter configurable lock delay, which lets us set a time delay after which the device auto-locks and needs a pin entry to be unlocked.
-- OR --
Having the device lock after the "Going to sleep" prompt.

### Justification

Currently the MiniBLE stays unlocked for a long time whenever it is not connected to USB (and has been previously unlocked). Coupling this with the fact that the MiniBLE has a "Display Credential" option means that this can be a security risk for whoever owns one. Having a configurable time-delay after which it auto-locks, or just having it auto-lock after the "Going to sleep" prompt, means that this risk may be somewhat mitigated.

The previous version did not have this issue as it could not be unlocked without it being connected to USB first and the inactivity auto-lock was much more granular.

### Workarounds

Remembering to lock the device by hand, each time.
Waiting 30+ minutes.

limpkin, 2021-02-10:
stupid question: what's the benefit of adding this feature when locking your computer locks the mini BLE?

GlitchOne, 2021-02-10:
If the device has been unlocked when it wasn't connected to a computer, then that feature cannot work, and the MiniBLE remains unlocked for a while.

limpkin, 2021-02-10:
@GlitchOne could you double check that? It shouldn't... unless the device is connected through Bluetooth to another master

GlitchOne, 2021-02-10:
I've just tested it with the following steps:
0. The MiniBLE is not connected to USB, or Bluetooth (it's disabled)
1. Wake MiniBLE from sleep.
2. Enter pin.
3. Choose login option.
4. Choose the first credential from the list.
5. Choose "Display Credential" as it is the only available option.
6. After it returns back to the main menu, I let it sleep.
7. I wait a set period of time.
8. I click the scroll wheel again, and am greeted with the main menu.

I waited 5, 10 and 20 minutes, and the BLE was unlocked after each timespan, and I could display credentials again. I'm guessing it auto-locks after 30, as that's the inactivity timelimit set in Moolticute, and will check it after this.

Considering that one may essentially unlock their BLE, display some credentials and leave their BLE unattended for 20 minutes, if someone comes by in those 20 minutes and clicks the scroll wheel, they are essentially able to dump passwords from the Mini without knowing the pin.

GlitchOne, 2021-02-10:
Update: Left it for 49 minutes, clicked the scroll wheel, was greeted with a main menu.

limpkin, 2021-02-10:
can you double check the inactivity lock is actually set? it should have locked in your case. I'll check on my side

GlitchOne, 2021-02-10:
I have. It is set to 30 minutes. I will set it to No Inactivity, then set it back to 30 Minutes, in case some config might've been leftover from an older firmware. Mind you, 30 minutes is still a lot in my opinion, and I liked the 5 minute inactivity timer I had set on the older Mini, hence the request.

GlitchOne, 2021-02-10:
Did the thing, waited 33 minutes, MiniBLE came back unlocked.

closed 2021-02-10 by limpkin

limpkin, 2021-02-13:
@GlitchOne did you try the updated timeouts? :)

GlitchOne, 2021-02-14:
Yes, @limpkin, I have and they are exactly whay I needed. Thank you!

commit 56443ca3 (2021-02-10): new inactivity timer allowing minute granularity, fix #222