mooltipass/minible #224: Keyboard layouts order in the Mooltipass settings

elendil-software, 2021-02-26:
### Missing feature

The keyboard layouts are ordered alphabetically (like in Moolticute). Currently they are ordered by OS non-MacOS, then MacOS and for each OS they are ordered alphabetically.

### Justification
It would be an acceptable workaround to this issue #130 by reducing the "scroll distance" (for those who use the same language on different OS devices)

MaplesThroughWindows, 2022-03-07:
This comment relates to using the Mini BLE over USB, not BLE.

Reducing Scroll Distance is a valid concern.  The Scroll Wheel is the part most likely to fail; and navigating the Keyboard Layouts list can take a while and can become tedious.  When moving between platforms with different Keyboard Layouts (I do this more than I thought) the current Keyboard Layouts setting is too cumbersome.  "Scroll distance" between macOS English and other non-macOS English layouts can be as many as 25 entries.  

**Alternative Suggestion:**

Give users the option to save Favourite Keyboard Layouts, and make this the first setting users see when entering the Keyboard Layouts menu.  After setting this up, this should reduce "scroll distance" considerably for users such as my self.

elendil-software, 2022-03-07:
I like the suggestion !

Hypoon, 2023-03-31:
I'd also appreciate this change, but for a different reason. When choosing a layout, it's not obvious that MacOS, Chrome, and Linux layouts even exist. Users shouldn't be expected to think, "Oh, this layout fits my region perfectly! Let me scroll through the entire remainder of the list just to see if there's an even better fit...".