mooltipass/minible #229: smc_decode_win.py script failing to decode passwords

Algid, 2021-06-09:
### Expected behavior
Script should decode passwords from database export.

### Actual behavior
After inputting my pin code i get the following:
```
Traceback (most recent call last):
  File "smc_decode_win.py", line 260, in <module>
    print(decrypted_backup_file[5][service_id]["name"] + ":", decrypted_backup_file[6][j]["name"], "/", decoded_password)
UnicodeEncodeError: 'utf-8' codec can't encode character '\udd11' in position 36: surrogates not allowed
```

### Step by step guide to reproduce the problem
1. Run smc_decode_windows.py: `python smc_decode_win.py`
2. Select dump using file picker
3. Enter `I understand`
4. Enter pin code
5. Error logged as seen above (log does include service & username for first credential, but no password)

### Operating System
- Windows 10

### Versions
- Python 3.6.8
- pycryptodome 3.10.1
- pyscard 1.9.9

### Full redacted log
```
C:\Users\Louis Collins\Downloads\minible-master\scripts\smc_decode>python smc_decode_win.py

Please insert your smartcard
connecting to ACS CCID USB Reader 0

> REDACTED
< REDACTED

Correctly changed card type to AT88SC102

> REDACTED
< REDACTED

Correct AT88SC102 card inserted
Card Initialized by a Mooltipass Device
Number of PIN tries left: 4
User Card Detected

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!  WARNING  !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!                                                                                                       !
! Using this tool effectively renders your Mooltipass device useless.                                   !
! After accepting the following prompt, your AES key will be fetched from your card.                    !
! Both your credential database and its decryption key will therefore be in your computer memory.       !
! If your computer is infected, all your logins & passwords can be decrypted without your knowledge.    !
! Type "I understand" in the following prompt to proceed                                                !
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Text input: I understand

Please Enter Your PIN: REDACTED

> REDACTED
< REDACTED

> REDACTED
< REDACTED

Correct PIN

> REDACTED
< REDACTED

AES key extracted: REDACTED
Decrypting file with CPZ REDACTED
Nonce: REDACTED

Decrypting memory file contents...
REDACTED: REDACTED / Traceback (most recent call last):
  File "smc_decode_win.py", line 260, in <module>
    print(decrypted_backup_file[5][service_id]["name"] + ":", decrypted_backup_file[6][j]["name"], "/", decoded_password)
UnicodeEncodeError: 'utf-8' codec can't encode character '\udd11' in position 36: surrogates not allowed
disconnecting from ACS CCID USB Reader 0
disconnecting from ACS CCID USB Reader 0
```

limpkin, 2021-06-10:
hello!
Is this with a mini ble or with a mini?

Algid, 2021-06-10:
Hello, the device is the new mini ble, versions if of interest:
Aux MCU: 0.58
Main MCU: 0.68

limpkin, 2021-06-15:
this looks like an invalid password.
I've updated the script to print warnings, could you try again?

Algid, 2021-06-15:
Thanks for getting back, I just gave it another go with the updated script and it seems that it did manage to list all of the accounts I currently have on the device along with usernames - although all of the passwords failed to parse with either:
```
REDACTED: REDACTED / Couldn't parse password for REDACTED  /  REDACTED
```
or:
```
REDACTED: REDACTED / INVALID_CHARS
```
where `INVALID_CHARS` is a string of broken characters (little boxes with question marks inside or just random chars).

The rest of the script output was similar to last time. Please let me know if you need any more information.

limpkin, 2021-06-15:
thanks again for reporting this to us :)

commit 42d4bd8f (2021-06-15): credentials decode script bug fix, fix #229
closed 2021-06-15 by limpkin