mooltipass/minible #230: Scroll large passwords with wheel

darmiel, 2021-06-10:
### Missing feature
Move the password back and forth using the scroll wheel.

### Justification
Currently, the password automatically moves back and forth during "Display Login". 
However, it is easy to lose track of where you are, especially with complex passwords.

Therefore, I would suggest that you can , no longer automatically.

MaplesThroughWindows, 2022-03-14:
I was just about to post the exact same request!

The functionality I would suggest is as follows:

1) When displaying a long credential of any kind (username OR password), then default to scrolling the credential back and forth automatically;
2) If the user intervenes by moving the scrollwheel, then the back-and-forth scrolling ceases and the user now directly controls the position of the credential with the scrollwheel.  
_Note_: This has the added benefit of resetting the display timeout, something I found myself using the scrollwheel for in any case - the back and forth of the display was slow enough on long passwords that the display would sleep before it bounced back to the part of the password I needed to read :)  I then had to re-access the credential from the Login list.
3) Enable a graphical indication of when the user has scrolled to the end of the credential, whichever end that may be.  This gives the user confidence that the scrollwheel has functioned correctly and is being acknowledged by the device.  I'm not sure what form this would take, given the limited capabilities of the display compared to a smartphone, where this kind of graphical feedback is common;
4) My immediate inclination is to suggest making the scroll wheel move the credential one character at a time.  This may or may not be the best way to go.  If a different method is chosen, I suggest that scrolling should always leave a portion of the credential that has been scrolled away visible to the user, as a reference point, similar to the way that Nano (the Terminal text editor utility) produces a portion of the previous window's text when navigating a line of text that moves off the screen.

rastagraffix, 2022-03-15:
MaplesThroughWindows writes:

> 3) Enable a graphical indication of when the user has scrolled to the end of 
> the credential, whichever end that may be.
>  [...]   I'm not sure what form this would take, given the limited capabiliti
> es of the display

Something like the unicode BLACK LEFT-POINTING TRIANGLE would likely do
the trick to indicate the end of the string in a scrolling situation?
And likewise its opposite-pointing partner at the left side of the
scrolled text.

> 4) My immediate inclination is to suggest making the scroll wheel
>    move the credential one character at a time.

I agree. Anything other than one character at a time is going to
make everyone's brain freeze up while they try to get their eyeballs
in sync again with the text.  Especially in some random gibberish
password string.

--lyndon

MaplesThroughWindows, 2022-03-16:
RastaGraffix writes:

> Something like the unicode BLACK LEFT-POINTING TRIANGLE would likely do
> the trick to indicate the end of the string in a scrolling situation?
> And likewise its opposite-pointing partner at the left side of the
> scrolled text.
> 

I think that would work well.

I'd suggest that these triangles flicker or flash if users continue to 
scroll when there is no more text to display, to draw attention to the 
fact they've reached the beginning/end of the credential.

These are the Unicode/UTF-8 codes I get for the suggested symbols:

BLACK LEFT-POINTING TRIANGLE - Unicode: U+25C0 U+FE0E / UTF-8: E2 97 80 
EF B8 8E
BLACK RIGHT-POINTING TRIANGLE - Unicode: U+25B6 U+FE0E / UTF-8: E2 96 B6 
EF B8 8E

jms1voalte, 2024-04-17:
I just found this issue when I was trying to figure out how to *make* it scroll in the first place. I have one particular password which is ~60 random (but type-able) characters long, and I can't figure out how to make it scroll side-to-side at all - which means that for this one password, I can't look it up on the screen to type it into a server, and because the only accessible USB port has been disabled in the BIOS, I can't log into it at all.

MiniBLE serial 1754, bundle 12 (I *just* found that 13 is available, I'll try upgrading and see if that changes anything.)