mooltipass/minible #233: miniBLE long scroll wheel press does not go back

kbembedded, 2021-06-16:
### Expected behavior
As documented, and with the mini, holding the button should go back. e.g. when entering PIN, press and hold to go back one digit to re-enter.

### Actual behavior
Pressing and holding the scroll wheel seems to advance it by one. It advances again if I then release the scroll wheel. This happens about 90% of the time. The unit seems to randomly on occasion actually interpret as a back press.

I have a second miniBLE available to me to test and this problem does not reproduce on that unit and holding the scroll wheel to go back is quite stable

### Step by step guide to reproduce the problem
See above

### Moolticute Version - If Involved
N/A

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
Mention if you are using either:
- Windows 10

### Workarounds
None

limpkin, 2021-06-17:
Hello,

this sounds like a hardware issue :/.
Can you send me a video of the isssue to support[at]themooltipass[dot]com ?

Thanks!

limpkin, 2021-07-01:
replacement unit shipped

closed 2021-07-01 by limpkin