mooltipass/minible #234: Hardware: Scroll wheel too slippery, design change recommendation

kbembedded, 2021-06-16:
The scroll wheel on the miniBLE does not have a stiff enough knurling on it to allow good finger contact. Its very rounded, and in cold, dry climates, a finger tends to slide across it rather than spin it. This requires more pressure to get the wheel to turn, which results in spurious wheel clicks.

If there are going to be further builds through the future and lifetime of the miniBLE, I highly recommend finding a different scroll wheel design or working with the manufacturer.

As a workaround, I'm going to try to do some light filing on the one I have here. If I find it to be successful I can report back with photos and more information for others.

limpkin, 2021-06-16:
Hello there!
We actually designed the plastic part of the scroll wheel ourselves (injection moulded).  
Could you send us a picture of your scroll wheel just to make sure there are no defects with it?

kbembedded, 2021-06-19:
See below. I don't think there is any kind of defect, the final result just does not have enough grip to catch on dry, calloused hands. It works well enough right after I wash my hands, on a humid day (its working better now that we have a heatwave coming through), but the past few days have been dry and cold and it was very difficult to spin it.

Like I said, I recommend a redesign to make the peaks and valleys more pronounced, with the edges where a valley meets a peak being sharper and not as rounded. I can slide my finger over the wheel and not move it at all, having to actually apply pressure to get it to spin. This is true on the other miniBLE unit I have access to as well. My mini on the other hand, its nearly impossible for me to move my finger across the wheel and not cause it to spin due to know knurled the surface is.

[image]
[image]