mooltipass/minible #235: OTP Auth

Properrr, 2021-06-16:
### Missing feature
OTP Auth
### Justification
We can implement the OTP on a device.

How to init codes?
-When an app shows a QR-code we can use Mooltipass GUI tool for select this QR code with an rectangle on a screen then desktop app can decode it and send initialized data to the device.
The web-browser plugin will able to detect a request of a OTP code the same way as credentials.

limpkin, 2021-06-16:
Hello!
Just to make sure I follow you: you'd like to see OTP and **not** TOTP on the device?

Properrr, 2021-06-16:
Hello @limpkin 
HOTP or TOTP would be great as well

limpkin, 2021-06-17:
but the mini BLE does handle TOTP :)

limpkin, 2021-06-17:
so should we rename this issue for HOTP?
however, if using HOTP.... syncing multiple devices together will create issues :/

0x0ptr, 2021-08-15:
@Properrr How widespread is the use of HOTP? I have not seen a site yet that uses HOTP but that does not mean it doesn't exists of course. Can you give me example of web sites you have encountered that support only HOTP (and not TOTP)? Any test sites that support HOTP?

Like @limpkin said this will break when you have multiple devices. The counter would need to be somehow carefully synch'ed between the devices....Also, when do you increment the counter? After MiniBLE did the autotyping? After displaying? Prompt to update the counter?

0x0ptr, 2021-09-01:
@Properrr Planning to close this issue within a week due to inactivity/no response.

Arffeh, 2021-09-01:
Funnily enough, I was eyeballing HOTP about 3 nights ago, and found the
ancient github issue before the MP-BLE existed.

I was looking at it due to some IdP products using it as an option.
(Keycloak comes to mind)

nlutsenko, 2021-09-07:
Coming from almost a decade usage of products by Yubico...
I can't be more than happy with MooltiPass, at the same time, the primary use case that I have is work-related, and for better or for worse, can't use U2F for it. Meaning that HOTP, the old and rusty, is what I use most of the time for 2-factor, and super surprised for it not being available.

One more data point:

Duo, as in https://duo.com.
These folks are multi-key 2fac provider, and they support most of Yubikeys + any OATH HOTP keys, but seems like not built-in others, resulting in pretty much dicouraging use of MooltiPass for it. (more here: https://duo.com/product/multi-factor-authentication-mfa/authentication-methods/tokens-and-passcodes)

---

Basically, literally can't wait to test out HOTP, how can we help to move this along?

0x0ptr, 2021-09-18:
Just to confirm that we are talking about the same thing. Your use case is mostly for logging in to work account and they only support HOTP - COUNTER based One-Time-Password (OTP) and NOT TOTP (Time-based One-time Password Algorithm)? And they are willing to provide you with the HOTP secret to use on any device [not managed by your IT department]? They don't just give you an "RSA SecurID" or a Yubikey (for U2F/FIDO) (for example) for the purpose of authenticating?

As discussed above there are multiple issues related to implementing this on Mooltipass MiniBLE as outlined above:
1) This will break when you have multiple devices. The counter would need to be somehow carefully synch'ed between the devices
2) When do you increment the counter? After MiniBLE did the auto-typing? After displaying the HTOP? Prompt to update the counter?

limpkin, 2022-03-22:
closing due to inactivity.

closed 2022-03-22 by limpkin