mooltipass/minible #236: Multiple BLE Devices

RChadwick7, 2021-06-16:
I was lucky enough to have two Mini BLE's arrive today. I paired the first one with a number of devices. However, I was unable to pair the second device. It almost appears as if the computer or phone can't see the second device, or sees it as the first device and gets confused. Is this a known issue?

limpkin, 2021-06-17:
that shouldn't happen.... can you disable the bluetooth interface on the first one and check that your computer sees the second one?

CubeCentral, 2021-06-17:
I am having the same issue.  I received both of my Mini BLEs yesterday and one of them I am able to pair with laptop, desktop, phone (it's a black one)  The other one I cannot pair with anything.  It is not seen at all.  The other (black) Mini BLE was nowhere near.  I've tried a desktop, laptop, and phone with no luck.  (it's a green one).  I was just looking for a tool to flash firmware on here, as I am out of ideas.

limpkin, 2021-06-17:
@CubeCentral is it a green device as well? could you send us a message at support[at]themooltipass[dot]com?

RChadwick7, 2021-06-17:
It appears one of mine is not working either. The first one shows up as soon as I turn Bluetooth on, but I turned it off, and turned it on on my second device, and even put it in pairing mode, but none of my devices ever see it. I got a Gray and Black. The Gray works, the Black does not.

CubeCentral, 2021-06-17:
> 
> 
> @CubeCentral is it a green device as well? could you send us a message at support[at]themooltipass[dot]com?

I have, and I have also spoken with you on IRC.  Thank you for your help!

limpkin, 2021-08-21:
that was a hw issue

closed 2021-08-21 by limpkin