mooltipass/minible #239: Potential issue with characters on iPhone

kbembedded, 2021-06-16:
### Expected behavior
Passwords successfully sent to iPhone via BLE or USB

### Actual behavior
In one isolated test (I cannot do more testing at this time, but would appreciate if someone else is able to replicate this), I found that using a pipe character '|' in a password, incorrectly was sent to an iPhone 12 (with latest iOS, not sure what it is) as a plusminus character '±'.

Keyboards were confirmed set up as "English US" for default/user, and "USA" for USB/BT keyboard in moolticute.

When connected to a Windows 10 PC, this was not a problem, the password was sent correctly.

When connected to the aforementioned iPhone, via USB or BLE, the password was sent correctly save for the '|' character in the password being incorrectly sent as '±'

In both cases, passwords were manually sent, i.e. not using any extensions to prompt for for credential approval.

If anyone else with an iPhone can reproduce this, it might warrant further investigation.

### Step by step guide to reproduce the problem
See above.

### Moolticute Version - If Involved
v0.46.3

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
- iPhone 12 iOS (unknown version, sorry)

limpkin, 2021-06-17:
did you select the (MacOS) variant of a given keyboard setting? :)

kbembedded, 2021-06-17:
That is a fantastic point. I will try to test this again with that and see if I can't reproduce it or make it go away!

kbembedded, 2021-06-18:
A new set of dummy credentials were created in the miniBLE which contained a pipe character in the password specifically. This was then re-tested on the same iPhone (See edits in original post for more detail) and produced the same plusminus char for both BLE and USB connections.

As per the keyboard settings, all original and "MacOS" variations of "USA" "US Extended" and "UK Extended" were tested. All MacOS and non-MacOS keyboard settings all incorrectly returned the plusminus character in the same way as outlined above.

elendil-software, 2021-06-19:
@kbembedded 
Are you using any third-party keyboard application ? If so, this could be related to this issue #215

kbembedded, 2021-06-19:
No third-party keyboards installed. However, the iPhone in question does have multiple language keyboards (Emoji, Greek, Japanese - Kana) enabled through iOS directly. These are not in use with this bug, the on-screen keyboard is default with default US iOS keyboard being used during testing.

gmarcusm, 2021-06-22:
I have a similar issue with iOS 14.6 over Bluetooth. BLE keyboard set to US Extended (MacOS), USB Keyboard to German (MacOS). In my case, the mooltipass is typing an extra space after a dash, so it is typing "- " instead of "-". The space is not part of the password, I verified that it is not present in the Credentials database.

limpkin, 2021-06-22:
So this could be one of two things, due to the way iOS is built :/
- especially in EU countries, some keys are dead keys (see https://en.wikipedia.org/wiki/Dead_key) and when combined with another character, produce a different key altogether (for example: ~ + a = ã). when such a dead key is detected, the mini ble forces a space character to produce the desired output (in this example: ~ + space + a = ~a). Strange thing is, "-" is NOT a dead key
- iOS and MacOS have a history of changing the user input depending on the application you type characters in :/

gmarcusm, 2021-06-22:
I am aware of dead keys. Also, MacOS have additional ones that are different from other platforms. In my particular case, switching the bluetooth layout from US Extended (MacOS) to USA (MacOS) fixed the issue. Maybe there is a wrong key modifier associated, that it thinks is a deadkey in this layout when it is not? with a "-" in the Mooltipass? 

In particular:
https://github.com/mooltipass/minible/blob/7f880dfa6829bcfbd7e5d388715435447e0ba3c9/scripts/keyboards/cldr-keyboards-37.0/keyboards/osx/es-t-k0-osx-extended.xml#L20

seems to have an issue.

kbembedded, 2021-06-22:
I have another datapoint relative to the issue with pipe that I originally reported.

### Operating system
- iPhone XS Max with iOS 14.6

Does _NOT_ reproduce the same problem with the pipe character as iPhone 12 Pro with iOS 14.6 originally tested on.

No third party keyboards installed on the XS Max either, but Apple keyboards installed are: English (US), Emoji, French (France), Greek, Japanese kana, and Spanish (Mexico). 

I'm trying to get another sanity check of the original problem with testing keyboards, power cycling things, etc. I'm also going to try to get the two together to see if the problem follows the phone or the miniBLE settings. 

(Apologies for the delays in gathering data, I'm coaching family and friends who have iDevices that I convinced to buy in to the miniBLE)

kbembedded, 2021-06-23:
After more testing, I think my original issue is actually a non-issue.

After the second iPhone was fine, I sat down with the original miniBLE/iPhone pair and:
- Rebooted the miniBLE
- Rebooted the iPhone
- Connected miniBLE to moolticute and say that "USA (MacOS)" keyboard was selected
- Set a different keyboard, saved, disconnected, reconnected, set "USA (MacOS)", saved
- Rebooted miniBLE
- Pipe character now prints fine to the original iPhone

I Have a feeling something went screwy with either device, or potentially the Settings from moolticute didn't take somehow. There might be some real deep bug hanging around that fact, but, for now, no more errant characters to mobile device.

wundercatXL, 2021-06-24:
Maybe what I see with the phantom extra character is a separate issue and I should file a new one.

gmarcusm, 2021-06-24:
maybe I should move the issue with the phantom character to a separate issue, seems unrelated to this.

limpkin, 2021-09-16:
hello all,
what's the current status on this bug? still there? :)

kbembedded, 2021-09-20:
I've not seen it from the family members that originally reported it. There certainly was some kind of issue, but doing a bunch of rebooting and re-setting of the keyboards used it seems to have all been resolved.

closed 2021-09-24 by limpkin