mooltipass/minible #241: Add additional text information to saved credentials

gmarcusm, 2021-06-17:
### Missing feature

It would be great from the usability standpoint if we can get a text string or textual notes attached to credentials. This key could be displayed on the mooltipass display after username and password is typed in, so that the user can complete the last step by itself.

### Justification

There are many services (specially in finance) that request a password, but then some form of text match (eg: please enter characters 4 and 9 from your secret key). 

### Workarounds

I have no way to work around this. I could probably add a file, but I neither match it with the credentials nor display it while logging in.

limpkin, 2021-06-17:
the database model allows a 3rd field to be added, but we're not making use of it in either moolticute or the extension.... I'm wondering if we should created associated issues on these repos as well:
https://github.com/mooltipass/moolticute
https://github.com/mooltipass/extension

gmarcusm, 2021-06-17:
I think there is something to be done, so "nothing we can do" is not the right tag.

An addition text field, either as 3rd field or note child node is possible.
To be added would be to display this field or note on the device display after the credentials are displayed.

I agree that this would require associated issues in moolticute and the extensions. Do you want me to create them?

limpkin, 2021-06-17:
I may misunderstood you... but the 3rd field is already there: https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/NODEMGMT/nodemgmt.h#L182  
Ohh... so you'd like to display it on the device? yeah then there's something to be done.
Yup, if you could create them and link to each other that would be great

limpkin, 2021-06-17:
should we therefore rename the issue for "3rd field on screen display"?

gmarcusm, 2021-06-17:
yes, renaming the issue sounds like a good idea if the field is already there.