mooltipass/minible #243: Randomize starting number when entering PIN

gmarcusm, 2021-06-18:
### Missing feature

The starting number for each digit in the PIN can be random instead of always zero.

### Justification

While very small, there is a possibility to learn/guess the PIN of a smartcard by hearing/snooping on the clicks of the wheel while a user is entering it (for example, with a microphone). Since the initial number is always zero, counting the clicks of the wheel gives a set of 16 possible values (4 digits, two possibilities, 2^4=16 values).

Granted, the possibility of this being used in the wild is very, very low. So this would be useful to only a handful of users. But it is still a possible attack vector.

### Workarounds

- 16 is still above the number of tries a user has for an incorrect PIN.
- Use different PINs in cloned cards.

limpkin, 2021-06-18:
hello there!
it's already there, in moolticute settings :)

closed 2021-06-18 by limpkin

gmarcusm, 2021-06-24:
Oh, doh. I missed it. thanks!