mooltipass/minible #245: WebAuthN failed login ends to Mooltipass freeze / reboot

limpkin, 2021-06-19:
### Expected behavior
After WebAuthN failed register/login, the device should still work as usual, like before the operation

### Actual behavior
After WebAuthN failed register/login, _canceled by the PC_ before any reply on the mooltipass, the mooltipass will freeze periodically after the prompt and show "#002 Contact Support" and sometime "#005 Contact Support" message.
The Mini BLE seems to loop indefinitely like that and some feature like WebAuthN or Memory Management Mode cannot be used from now. The Mini BLE will also restart if usb is unplugged (not always, but most of the time).
The only found way to come back to normal use is to restart the Mini BLE.

### Step by step guide to reproduce the problem
Bug reproduced on another Mini BLE/card pair but not tested on Windows. 
1. The Mini BLE is connected by usb with bluetooth disabled
2. Go to a WebAuthN enabled website like https://webauthn.io (but not limited to)
3. Register with WebAuthN and while the Mini BLE is asking you to save credentials, cancel the request on the PC
4. Wait Mini BLE remove the prompt or reply anything
5. The bug is now enabled

### Operating System
- Ubuntu - Vivaldi browser
- MacOS - Safari / Vivaldi

### Mooltipass Extension
- the chrome extension (with Vivaldi macos): 1.8.1 
- the Safari extension: Not used

### Mooltipass Application
- Moolticute v0.46.3

### Mooltipass Device
- The Mooltipass Mini BLE

limpkin, 2021-06-19:
@TLeM4 is bluetooth enabled when this happens?

TLeM4, 2021-06-19:
@limpkin As written in the step-by-step, i usually disable Bluetooth but with one more test with Safari/MacOs it's seems that doesn't change anything : the bug is still present.

limpkin, 2021-07-01:
@TLeM4 thanks a lot for reporting this issue to us... it turns out some funky USB choice on Ubuntu was causing a bigger issue on the Mini BLE.

commit 0dab7498 (2021-07-01): dedicated DMA buffers for RNG/FIDO2/BLE message, fix #245
closed 2021-07-01 by limpkin