mooltipass/minible #248: Minible Freezing After Manual Input From Device w/ Bluetooth Enabled

sonofapharmacist, 2021-06-21:
8<------------------------ BUG REPORT -----------------------------------------

### Step by step guide to reproduce the problem
1) Input password directly from selection menu on minible to Edge w/o the moolticute extension installed in Edge.  
2) After selecting the credential on the minible it asked if i wanted to use usb or bluetooth, I select usb.
3) Input credentials to PC fine.
4) Minible screen freezes on primary menu with Login selected.
5) Disconnect from USB, device screen turns off, device will not wake.
6) Plug in USB, device still frozen on primary screen.
7) Turned off bluetooth on paired device, repeat 5 and 6, same.
8) Try to interact at all from Moolticute, minible does not respond.

### Firmware Version
AUX MCU version:  moolticute does not register a version after this freeze
Main MCU version:  moolticute does not register a version after this freeze

### Operating System
Mention if you are using either:
- Windows 10 (multiple computers)

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-06-21:
wow that's a first... is this reproducible? what's your computer setup?

sonofapharmacist, 2021-06-21:
I can't reproduce presumably until the battery discharges fully.

I was on win 10 hp zbook. Connected through a 4 port USB switcher. This is
the work laptop. The laptop's USB controller has been overheating lately,
sometimes everything disconnecting or not working out right.

I tried connecting the frozen device to my other pc through a different USB
port but it doesn't respond.

limpkin, 2021-06-22:
we're fairly sure the firmware update we'll be releasing in the coming weeks will fix that :)

limpkin, 2021-07-03:
@sonofapharmacist  could you get in touch with us (support at themooltipass dot com) so we can send you a firmware update to test?

limpkin, 2021-07-12:
how the new firmware update performing? should we close this?

closed 2021-08-01 by limpkin