mooltipass/minible #249: scroll wheel click bounce issue

depperson, 2021-06-22:
### Expected behavior
When I click the scroll wheel one time, only one character should be entered at the lock screen.  When I hold the scroll wheel down, it should reliably function like a backspace, allowing me to correct an incorrectly entered character. 

### Actual behavior
While entering my unlocking PIN, when I click the scroll wheel one time, it frequently registers two clicks instead.  I blocked one card accidentally due to incorrect PIN entry.  I am also unable to use the "backspace" reliably when holding the scroll wheel clicked down, sometimes nothing happens until I release it, which is registered as a single click instead.  I have received two minible kits and only one of them has this problem.  The problem unit is very reliably misbehaving, I would say ~50% of single clicks are entering two digits at the lock screen.  This behavior makes it impossible to repeatably enter a PIN code other than 0000.  When the problem is observed, three clicks results in 0000 entered because one of the clicks was doubled. 

### Step by step guide to reproduce the problem
1. Create new user with blank card. 
2. Configure 1111 as the key.  
3. Remove the card and enter the PIN.
4. If a digit is skipped it will be 0 by default, causing incorrect PIN entry. 
5. Repeat #3 until #4 is observed, it should only take a few tries. 
6. In the middle of the PIN entry, try to use the click+hold backspace function. 
7. If #6 works then try it again a few more times. 

One device this always fails, while the other device works fine. 

### Moolticute Version - If Involved
N/A

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
- MacOS  10.13.6

### Mooltipass Extension
- Not related

limpkin, 2021-06-22:
I'm afraid that's a hw issue with your device. Could you get in touch with me at support[at]themooltipass[dot]com so I can get your device replaced?

limpkin, 2021-06-23:
closing as hw issue.

closed 2021-06-23 by limpkin