mooltipass/minible #250: Mini BLE freezing on manual credential input with no connected device

KDederichs, 2021-06-22:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
It doesn't freeze when trying to manually input credentials when no device is connected

### Actual behavior

It hard freezes, I can't seem to do anything, screen stays on, button animation is frozen, controls are not responding

### Step by step guide to reproduce the problem

1. Connect BLE via bluetooth
2. let it sit for a few hours so that it disconnects automatically
3. manually select login
4. select the first item on the list
5. confirm login input

### Firmware Version
AUX MCU version:  what ever it shipped with
Main MCU version:  what ever it shipped with

### Operating System
Mention if you are using either:
- MacOS  

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-06-23:
fairly sure https://github.com/mooltipass/minible/commit/b26bf79c748382d71d764a878afd2d4a435ad6d9 fixes it, but keeping it open in case.

limpkin, 2021-07-03:
@KDederichs could you get in touch with us (support at themooltipass dot com) so we can send you a firmware update to test?

KDederichs, 2021-07-04:
Doesn't seem to crash anymore but it's also not inputting stuff via Bluetooth (using the steps outlined above).
Don't know if that's related but it's an improvement :)

limpkin, 2021-07-07:
@KDederichs you're saying you can't use manual credential recall? :/

KDederichs, 2021-07-07:
No, it's still working on cable and I assume after after reconnecting the BT connection?
Just right after the scenario that crashed it previously it didn't seem to transmit data via BT

limpkin, 2021-07-08:
could you restart your computer? there's a well known issue on macOS that after first pairing, characters can't be typed :/

limpkin, 2021-07-12:
how the new firmware update performing? should we close this?

KDederichs, 2021-07-12:
yeah seems to be working fine on my end, I didn't experience any freezing yet so I'll just go ahead and close this one :)

closed 2021-07-12 by KDederichs

limpkin, 2021-07-12:
nice :)