mooltipass/minible #252: [OSX] Mini BLE "stole" keyboard status?

KDederichs, 2021-06-25:
I'm not 100% sure that that's what happend but it might be:

### Expected behavior
You can have a Magic Keyboard and a BLE at the same time and they never will interfer with one another

### Actual behavior
I had this funny problem today where the Mac didn't recognize the keyboard on boot.
After plugging in a USB keyboard and logging in I saw that it didn't show up in the list of connected bluetooth devices but it did list the Mini BLE as connected via Bluetooth.
So I assume the BLE might have taken the status of primary keyboard or something?

Like I said I'm not 100% that's the cause but I'll just mention it here just in case.

### Step by step guide to reproduce the problem
1) Connect BLE and Magic keyboard by Bluetooth
2) Shut down a few times
3) Maybe it'll happen?

### Firmware Version
AUX MCU version:  factory default
Main MCU version:  factory default

### Operating System
Mention if you are using either:
- MacOS  

Like I said, not 100% sure it's related, anyone having a similar experience?

spoelstraethan, 2021-07-01:
Apple does seem to initialize devices in a weird way, are you using a Mac Mini or iMac or Mac Pro?

Since the MiniBLE appears as an HID device it might be possible that Apple accepts it as a keyboard and connects before other devices, but I wouldn't expect it to ignore an also paired Magic Keyboard or a USB keyboard just because it has an HID device already.

KDederichs, 2021-07-01:
Happened on an iMac (2019)