mooltipass/minible #253: Knock detection stopped working after long uptime.

awilkins, 2021-06-25:
### Expected behavior

- Knock detection works

### Actual behavior

- After a few weeks of operation, knock detection stopped working for credential prompts
- I know the accelerometer still works, because the device still wakes when picked up

I have the Mooltipass plugged into my home workstation and don't unplug it when stopping work for the day, the port retains USB power on shutdown, so I think it's likely it's not fully rebooted since I first received it.

### Reproduce

This only happened after a couple of weeks of continuous operation, so not happened more than once.

### Workaround

What worked was setting "Switch Off Device After USB Disconnection", pulling the USB plug, and rebooting. This restored knock detection.

### Firmware Version

AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System

- Debian Based distribution (Ubuntu 20.04)

limpkin, 2021-06-25:
wow that's going to be an interesting bug to diagnose.... 
are you sure it doesn't happen after a few days?

awilkins, 2021-06-25:
I'll leave my MiniBLE plugged in again and reset the USB power option - next time it happens I'll check the time against this ticket. For my memory, this started yesterday.

For my money (and without having looked at the code), it's either down to some variable holding a timestamp value overflowing... or maybe a memory leak.

filipemagomes, 2021-06-29:
I have the same issue.
But I use MP connected by bluetooth.

The workaround for me is just press the wheel for 15 secs to restart the MP.

The device don't wakes when picked up.
I always charge the MP on a smartphone charger not connected to the computer.
I have two devices and the issue happens on both.

awilkins, 2021-06-30:
Had to reboot as described above yesterday, device unresponsive (to anything, including wheel movements) so the clock is reset on "how long does it take for the knock to not knock".

limpkin, 2021-07-01:
so it does look like some unidentified event is causing the accelerometer to stop transmitting data, therefore creating a cascade of events (in some case a 002, in another this "soft bug"). currently investigating

limpkin, 2021-07-03:
@awilkins @filipemagomes could you get in touch with me to test a firmware update? support(at)themooltipass dot com

limpkin, 2021-07-12:
how the new firmware update performing? should we close this?

filipemagomes, 2021-07-12:
Hi Mathieu,

I'm still on vacation. I used it a few times and I didn't have any
problems, but for the week I'll test better.

awilkins, 2021-07-14:
7 days after install, no problems so far.

filipemagomes, 2021-07-19:
Hi,

The problem remains when in use with bluetooth. I made a video, if you want I can share it.

limpkin, 2021-07-27:
@filipemagomes did you update to bundle v3? 
https://www.themooltipass.com/updates

filipemagomes, 2021-07-27:
Hi, I didn't know there was a V3 bundle. Where can I check for new releases in the future?

But answering the question, I can't upgrade to bundle v3. I have two devices on bundle v2, and moolticute give same error.

Using the link above, and make "Fetch Update File", i'm geeting a file with name update_serialnumber_from_bundle_0_to_3_uploadpassword.img 

I don't know if the file is the correct one. The file shouldn't be something like update_serialnumber_from_bundle_2_to_3_uploadpassword.img?

limpkin, 2021-08-01:
www.themooltipass.com/updates will stay there for a long while I think...

@filipemagomes can you send me a private email so I can send you the right upload passwords for the 0_to_3 files?

filipemagomes, 2021-08-05:
Hi,

I've already updated to bundle v3, but when I'm connected via bluetooth the problem remains.

limpkin, 2021-08-21:
@filipemagomes @awilkins how's your experience so far?

filipemagomes, 2021-08-23:
Hi, it's much better now.
Sometimes when I unlock mooltipass it doesn't work the first time I try to authenticate on some site. But as soon as I authenticate manually by pressing the wheel then it always works.

limpkin, 2021-08-31:
ok! i'll go ahead and close the issue then :)

closed 2021-08-31 by limpkin