mooltipass/minible #258: Keyboard Layouts seem to be randomly reset to Belgium French on both USB and Bluetooth

ubergeek77, 2021-07-01:
### Expected behavior
Keyboard layouts, once changed by the user, should never change to the default value (Belgium French), which is at the top of the list. The layout settings should be saved permanently, even after complete battery drain or long periods of non-use.

### Actual behavior
At random, for unexplained reasons, the keyboard layout for both interfaces (USB and Bluetooth) seems to be resetting themselves back to Belgium French. I use the USA layout for my passwords, so this causes incorrect inputs unless I manually change the layouts back to USA once more.

### Step by step guide to reproduce the problem
I cannot provide a step by step, as this is happening at random times for unexplained reasons. However, I can provide context for my usage scenario and insights that may help narrow down the problem:

- This has happened 3 times so far since I've received the device.
- I use only 1 profile, I have no other profiles on the device. In other words, this is not due to the setting being different on another profile, because I do not use another profile.
- This profile was migrated from a Mooltipass Mini. I use a newer style card that was cloned from my original Mini card.
- I have never changed the keyboard layout using Moolticute. I always change it on the device.
- I rarely use Moolticute, I have not had to add more credentials to my device in over a week.
- I originally thought this issue was caused by the battery fully depleting, which was when I first observed this issue. Since then, I made sure to always keep the device charged, and still today the layout was changed back to Belgium French, despite the battery being fully charged at all times.
- This seems to happen after extended periods (12-24h+) of not using the device at all. When using the device throughout the day, USA remains as my chosen layout.

### Moolticute Version - If Involved
v0.46.3 (however I do not believe it is related)

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
Not applicable

### Mooltipass Extension
Not applicable

---

Please let me know if there is any additional info I can provide that may help in narrowing down the problem.

limpkin, 2021-07-01:
hmm that's a fun one... could you update moolticute to the latest and try again?

ubergeek77, 2021-07-01:
Alright, I updated Moolticute, and once it opened again, it opened to the "Device Settings" tab, and something immediately caught my eye...

[image]

I don't really spend any time in Moolticute, so I didn't catch this (or even know this was a feature). It seems Moolticute had defaulted itself to Belgium French, and also had the "Enforce" checkboxes checked by default. I certainly didn't select these as I don't even speak French.

However, this must have been it. I do use my PC to charge my device, and when it's not charging, I have it disconnected to use wirelessly with my phone. This would explain the "randomness," it was just Moolticute doing it in the background.

I'll go ahead and close this then! But just a casual suggestion, if this was not changed from v0.46 -> v0.50 those "Enforce" checkboxes were probably not intended to be enabled by default.

Thanks for your time!

closed 2021-07-01 by ubergeek77