mooltipass/minible #260: clone card to non-empty card lead to two invalid cards

GeekAnnoyed, 2021-07-04:
Sorry if this is a duplicate bug

8<------------------------ BUG REPORT -----------------------------------------
### Expected behaviour
Clone card 1 to card 2 
Insert can 1 read it after entering PIN and clone over card 2

### Actual behaviour
Card 1 is cloned Insert non-blank card 2, receive error about card 2 not being empty
Neither card 1 nor 2 works any more

### Step-by-step guide to reproduce the problem
This is from memory, as I daren't risk losing all my passwords again.

Insert card one, fill with passwords.
From the BLE menu Clone
enter pin to confirm
enter pin for new card twice
insert card two, receive error  about card two isn't empty
reinsert card one, get error about something like card is invalid.
Remove card one, insert card two. Card two is also invalid

unable to restore backup DB as "this backup was for a different user"

### Moolticute Version - If Involved
v0.50
### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
Mention if you are using either:
- Windows 10

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-07-04:
Hello!
I'm fairly sure this must be something else, as we have mechanism to actively prevent this use case...

GeekAnnoyed, 2021-07-04:
I will test it again when I've got a 3rd card to test with. 
This isn't the first time it has happened, It happened the first day I had my BLE I put that down to user error/excitement :)

limpkin, 2021-07-04:
I may have an idea of what may have happened, looking at this line of code: https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/LOGIC/logic_gui.c#L143
When reinserting the original card, you're getting "invalid card" in any case.

GeekAnnoyed, 2021-07-04:
Yes I was. exactly that

limpkin, 2021-07-04:
hmmm in that case we should detect if the original card was inserted back

commit 8232df1d (2021-07-12): fix #260
closed 2021-07-12 by limpkin