mooltipass/minible #263: Mini BLE "Type Password" feature adds space after character ^

TheGreatBeginner777, 2021-07-08:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Fill in password field with password as saved in Credentials tab in moolticute app. 

### Actual behavior
If password have character(s) ^ upon automatic typing through Mini BLE it will add space after each ^ character.
Keyboard Layout in this case is set to Slovenian .

### Step by step guide to reproduce the problem
1. Within moolticute app in tab Credentials setup new "bogus" service + username + password
2. Password must have at least one character ^
3. Save new credentials to device
4. Try to type (fill in) the credentials from Mini BLE in text editor of your choice.

### Moolticute Version
v0.50.0
### Operating System
- Other Linux distribution: Ubuntu 20.10

### Mooltipass Device
- The Mooltipass Mini BLE

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-07-08:
So this is going to be an interesting one :)
- on windows: what happens if you only press the key associated with ^ ? does it type it or do you need to add another character?
- on linux: could you do the same?

TheGreatBeginner777, 2021-07-08:
> I've had similar issues with Linux, would you please try testing again after setting the keyboard to the "(MacOS)" variant and see if that has any effect? Doing this, with the USA keyboard, worked for me, and the layout is still compatible with other OS's.

What I did is set in Moolticute USB Keyboard Layout: USA (Mac OS) and on Ubuntu keyboard: set to English (US)->QWERTY. 
Now I do not have reported issues. 
What I do not like is that I moved from QWERTZ to QWERTY and a lot of specific alphabet signs are now missing (meaning I am not able to write them down with keyboard combination).

TheGreatBeginner777, 2021-07-08:
> So this is going to be an interesting one :)
> 
> * on windows: what happens if you only press the key associated with ^ ? does it type it or do you need to add another character?
> * on linux: could you do the same?

Windows 10. Keyboard: QWERTZ.
If I need to import character ^ I need to press ALTGR+3+3 (meaning I have to press number 3 two times and it will give me two character ^^)
But on Windows 10 if I try to fill in the password everything is working out of the box like it should be. I don't have problems like described above.

Ubunut 20.10. Keyboard: QWERTZ
If I need to import character ^ I need to press ALTGR+3.

limpkin, 2021-07-08:
I think that settles it: we have a different behavior when in windows or ubuntu.
@TheGreatBeginner777 what if  you wan to type â or ê ?

TheGreatBeginner777, 2021-07-08:
Windows 10.
I have to change keyboard layout to International and then use keyboard combinations (e.g. â = ^ + a)

Ubuntu 20.10.
I have to change keyboard layout to US, international with dead keys and I also have to set up a compose key (e.g. left alt) and then use keyboard combination (e.g. â = ^ + a)

limpkin, 2021-07-08:
does ubuntu have compose keys disabled by default?

TheGreatBeginner777, 2021-07-08:
it was on my end. so I had to enable it and set preferred keyboard key

limpkin, 2021-07-13:
@TheGreatBeginner777 sorry for the delay in answering... does that mean this issue is only valid when you change ubuntu default settings?

TheGreatBeginner777, 2021-07-13:
I would say the other way around. If I'm on my default settings (default keyboard layout, disabled compose key) then this issue is present. 

one minor workaround that I made to avoid this one is to define password profile in moolticute without character ^ which is causing me problems at the moment

limpkin, 2021-07-13:
that's extremely interesting.... correct me if I'm wrong, but that does effectively mean that by default, a given user wouldn't be able to type â ê ë etc on ubuntu with a mooltipass?

TheGreatBeginner777, 2021-07-13:
that is correct.

maybe I'm doing something wrong in here. but with my default settings I'm not able to type those letters since compose key is disabled so I can't type the keyboard combination and get those mentioned letters.

limpkin, 2021-07-13:
hmm that's quite a disappointment...

awilkins, 2021-07-13:
Ubuntu 18.04 and 20.04

Arrived at this ticket after noting that :

Bundle 2:

- UK keyboard layout
- Password contains a `^`
- HID types a `¾`
- Creds Management successfully fetches the `^` (I'm guessing the extension uses this API also from the prompt)

I only have UK input methods installed. Compose key is off on both. I do have extra layout options but they're all to do with non-char keys (ESC, CAPS and SUPER).

- `0101110` is `^`
- `1011110` is `¾`

limpkin, 2021-07-18:
actually we decided to release bundle v3 update now. Could you send me a message to support(at)themooltipass(dot)com to get your firmware and check if this has been fixed?

limpkin, 2021-07-27:
closing this issue in a few days if no answer soon :)

TheGreatBeginner777, 2021-07-27:
> actually we decided to release bundle v3 update now. Could you send me a message to support(at)themooltipass(dot)com to get your firmware and check if this has been fixed?

hi @limpkin 
email sent to this email address.

TheGreatBeginner777, 2021-07-28:
hi @limpkin 
updated minible to Bundle version:3; Aux MCU: 0.64; Main MCU: 0.75.
With default settings from 1st post (minible set to USB Keyboard Layout: Slovenia) I'm experiencing the same behaviour like described in 1st post. I must be doing something wrong in here so I gave up on and decided to use password profile without this headache of ^ symbol. I'm not that experience as awilkins to give you better input to this error I'm experiencing.

limpkin, 2021-07-30:
It's an oversight on my side....
Just to make sure I fully understand, using a slovenian keyboard:
- ^ + a = â on windows
- ^ + a = ^a on ubuntu ?

TheGreatBeginner777, 2021-08-02:
hi @limpkin 
correct. that is behaviour on my side since in ubuntu compose key is disabled by default.

limpkin, 2021-08-02:
hey @TheGreatBeginner777  :)
And if you enable the compose key but not use it, the behavior remains the same, right?
The only way to have the same behavior between windows & ubuntu is to enable the compose key, and press the compose key before typing ^ and a, right?

TheGreatBeginner777, 2021-08-02:
> And if you enable the compose key but not use it, the behavior remains the same, right?
correct.

> The only way to have the same behavior between windows & ubuntu is to enable the compose key, and press the compose key before typing ^ and a, right?
correct.

limpkin, 2021-08-03:
wow... that s*cks.
Do you know if it's the same for other european keyboard layouts?

TheGreatBeginner777, 2021-08-04:
i really don't know that.

limpkin, 2021-08-21:
I think we should investigate if that's the case before we start adding an option that differentiates Windows layouts with Linux ones :/

limpkin, 2021-08-25:
I've run a few tests on an ubuntu 20.04 LTS (stock) and don't have the same behavior for slovenian (and french) :/. Effectively, i have the same "approach" to ^ between windows and ubuntu :/

TheGreatBeginner777, 2021-08-25:
then this must be something on my side (OS itself or keyboard settings). for now I will avoid adding ^ to my password until I figure out what is the issue on my side...
thanks for the effort on this subject!

limpkin, 2021-08-28:
it's one of these things that tickle my brain :).
did you enable the compose key feature?

TheGreatBeginner777, 2021-08-30:
I applogize for late replay.

yes, compose key is enabled (mine is set to "Right Super").

limpkin, 2021-08-31:
ohhhh that must be it then! I need to add an option to take into account operating systems with compose key enabled

limpkin, 2022-03-21:
so I'm slowly getting back to this issue....
@TheGreatBeginner777 can you confirm that enabling the compose key effectively disables all standard transforms (typing ^ then a to generate â for example)?

TheGreatBeginner777, 2022-03-23:
hi @limpkin 
In my setup that is the case. If I want to type â, with compose key enabled, my keyboard combination is: 
compose key+alt gr + 3 + a 
As you can see I can type this form of letter â but it takes 4x keyboard press to do so. Please correct me if I'm doing anything wrong.

limpkin, 2022-03-28:
No worries, I'm trying to better understand compose keys and how used they are :)