mooltipass/minible #265: Ensure FIDO spec and Chrome security key requirements are met

spoelstraethan, 2021-07-08:
Important and concerning section highlighted in the middle of the quote below.

https://www.chromium.org/security-keys

> Site attestation requirements
> 
> Chrome’s users have an interest in ensuring a healthy and interoperable ecosystem of Security Keys. To this end, public websites that restrict the set of allowed Security Keys should do so based on articulable, technical considerations. They should regularly update their set of trusted attestation roots that meet their policies (for example, from the FIDO Metadata Service) to ensure that new Security Keys that meet their requirements will function.
> 
> Chrome expects Security Key manufacturers who have passed FIDO certification to keep their attestation metadata updated in the FIDO Metadata Service. 
> 
> If Chrome becomes aware that websites are not meeting these expectations, we may withdraw Security Key privileges from those sites. If Chrome becomes aware that manufacturers are not maintaining their attestation metadata we may choose to disable attestation for those devices in order to ensure a healthy ecosystem.
> 
> Attestation Batches
> 
> Security Keys are able to present an attestation certificate when registering with websites. These attestation certificates are intended to provide assurance to the site about the level of protection provided by the Security Key. However, they are also a privacy concern if the certificate is unique to a small number of Security Keys.
> 
> **Google Chrome expects Security Keys to meet or exceed the FIDO guidance in this respect and thus have attestation certificates that cover at least 100,000 devices. When starting a new batch, manufacturers may temporarily have fewer than 100,000 devices using a given certificate. But, in the long run, Chrome expects at least 100,000 to share a certificate. However, Chrome expects at least 100,000 devices to share each new certificate before a new batch is begun.**
> 
> **If Chrome becomes aware that certain Security Keys are not meeting these requirements we may choose to disable attestation for those devices in order to protect the privacy expectations of our users.**
> 
> (This does not apply to registrations where Chrome has signaled to a Security Key that it is operating in an enterprise context as detailed above—certificates may be individually identifying in this case.)
> 
> If Chrome becomes aware that devices are overly identifiable via other aspects of the protocol we may choose to cease allowing registrations, or to cease all interoperation with these devices for the same reason.
> 
> User presence tests
> 
> Some Security Keys are required to confirm user presence before performing certain operations. This is expected to prevent silent registration and silent authentication, both of which could violate our users’ privacy expectations. Chrome requires Security Keys to conform to the relevant specifications and to effectively implement these checks when required.
> 
> If Chrome becomes aware that certain Security Keys can be abused to allow silent registration or silent authentication we may choose to cease interoperating with these devices in order to protect the privacy expectations of our users.
> 
> User verification tests
> 
> Some Security Keys are capable of verifying the presence of a specific person, for example with a fingerprint sensor or PIN entry. Chrome expects these Security  Keys to effectively protect this personal information.
> 
> If Chrome becomes aware that certain Security Keys are disclosing personal information, such as PINs or fingerprint hashes, then we may choose to cease interoperating with these devices in order to protect user privacy.

spoelstraethan, 2021-08-10:
It appears Microsoft has a process for getting into their "certified" FIDO device list that could also be useful for making the Mooltipass BLE more well know.

https://docs.microsoft.com/en-us/azure/active-directory/authentication/concept-fido2-hardware-vendor

0x0ptr, 2021-08-15:
@spoelstraethan 

> But, in the long run, Chrome expects at least 100,000 to share a certificate. However, Chrome expects at least 100,000 devices to share each new certificate before a new batch is begun.

I don't think it is possible for MiniBLE to accomplish this. There are multiple issues. 1) I don't think there will be 100000 MiniBLEs produced... 2) We only support self-signed certificate. To have a manufacturer certificate requires hardware changes to store the private key securely on the device. There is no way to do that now.

My1, 2023-02-18:
also I would say this paragraph only applies to FIDO devices that use attestation certs in the first place. the self-attestation, instead of a fixed attestation certificate and key always uses the credential private key to sign the attestation claim, which does not really have any assurances compared to no claim at all but at least is able to provide info.

Basically instead of just saying nothing about itself, or saying "I am a Yubikey and I have a certificate from Yubico to prove that" the Mooltipass just goes "I am a Mooltipass, but not really anything to prove that, trust me bro"