mooltipass/minible #270: minible types wrong character

schlomie, 2021-07-13:
### Expected behavior
minible should type all characters in a password exactly as persisted and displayed

### Actual behavior
some characters (the backslash `\` in this case) are typed out as a different character, (less than `<` in this case).   It is unknown whether only the `\` is affected.

This happens on both USB and BLE

### Step by step guide to reproduce the problem

1. Create a test credential.
2. Give it a password which contains a backslash, e.g. `test\`
3. Confirm the backslash is saved on minible by displaying on the LCD and/or through moolticute
4. Have minible type out the test password
5. Confirm what is typed out is `test<`

### Moolticute Version - If Involved
Not directly related, I don't think, but built from git, master branch, commit 045f9a4

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
Void Linux - 5.12.14

### Other Useful Information
Device Default Language - English US
Current User Language - English US
Bluetooth Keyboard Layout - USA
USB Keyboard Layout - USA

schlomie, 2021-07-13:
Related - #263 - might be a duplicate?

I modified the keyboard layout to USA (MacOS) and it typed out the `\` as expected.

limpkin, 2021-07-13:
hello! did you select the keyboard layout that matches yours? :)

schlomie, 2021-07-13:
> hello! did you select the keyboard layout that matches yours? :)

I did.  When USA is selected as the keyboard layout (which is correct) it types out the `<`.   Using the suggestion in #263 to use the MacOS variant of the USA layout, the `\` is typed out correctly.

limpkin, 2021-07-14:
I'm fairly sure this is due to the way the US layout is setup on void linux :/

schlomie, 2021-07-14:
Hmm.  Well, the plot thickens.   The problem isn't exclusive to void linux.  I was able to replicate the same on:

- Arch
- Debian
- Ubuntu
- Mobian (pinephone)

But interestingly - it worked correctly on LineageOS on an android phone.

It is prudent to note that similar HID devices, with similar feature sets <sup>1, 2</sup> work as expected on all of these Linux platforms using the same US Keyboard layout.

<sup>1</sup> [OnlyKey](https://onlykey.io/)
<sup>2</sup> [InputStick](http://inputstick.com/) via keepass2android plugin

limpkin, 2021-07-15:
Ok I'm starting to think this is a CLDR related issue.
We're basically parsing this file to know how to type \ : https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-t-k0-windows.xml#L46

I'm therefore fairly sure we should take the mapping for the next \, mapped to <map iso="B00" to="\"/> 

If you agree, i'll implement the change and close that issue

schlomie, 2021-07-16:
The comments in the XML shine a bit of a light on the issue.   The CLDR seems to map to a UK (ISO) variant of the QWERTY keyboard, instead of the US (ANSI) layout.  Does CLDR only have mappings for ISO layouts?

https://en.wikipedia.org/wiki/British_and_American_keyboards

If you notice, the US layout does not have the backslash to the right of the apostrophe [46](https://github.com/mooltipass/minible/blob/96c422fe5a104791c5c6f368c54152729ec6dc81/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-t-k0-windows.xml#L46), nor any where near the Z [47](https://github.com/mooltipass/minible/blob/96c422fe5a104791c5c6f368c54152729ec6dc81/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-t-k0-windows.xml#L47)

The UK keyboard does have the backslash/pipe key directly to the left of the Z.   So, I think your observation and suggestion will indeed work - but what other inconsistencies might we find?

Anyway - I'd be willing to try a debug build on my new minible if you wanted to try a build without touching the master branch.  Otherwise, I'm good with this change - provided it works. :wink:

limpkin, 2021-07-16:
fixed and tested :). this will be included in bundle v3 release in 2 to 3 weeks :)

limpkin, 2021-07-18:
actually we decided to release bundle v3 update now. Could you send me a message to support(at)themooltipass(dot)com to get your firmware and confirm that this has been fixed?

limpkin, 2021-07-19:
confirmed to be fixed :)

schlomie, 2021-07-19:
Confirmed!  Thank you!

schlomie, 2021-07-20:
Well bugger.  It appears the pipe character is also affected by this.

Same as above with the backslash, the pipe `|` is storing and displaying correctly, but is being typed out as greater-than `>`

Should this bug be reopened, or would you like for me to open a new issue?

schlomie, 2021-07-20:
Similar - swap [97](https://github.com/mooltipass/minible/blob/ed65aaf3c3899406c5ceba7d6b3310ab7598287e/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-t-k0-windows.xml#L97) with [98](https://github.com/mooltipass/minible/blob/ed65aaf3c3899406c5ceba7d6b3310ab7598287e/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-t-k0-windows.xml#L98) ??

limpkin, 2021-08-24:
hello there!
somehow I didn't get notified of this issue... I'm getting an issue where I can't type * on french keyboard.

limpkin, 2021-08-24:
I'm sooooo dumb.

closed 2021-08-24 by limpkin

schlomie, 2021-08-24:
Heh.  Those happen.   Anyway - my problem post v3 was resolved.  Somehow moolticute was configured to switch to a UK layout upon connection.  Dunno how that happened, but since being reliably on the US layout, it's been flawless.

Terrific little device.

limpkin, 2021-08-24:
thanks for the kind words :)

commit 2a80c225 (2021-07-16): correct backslash character for US layout, fix #270
commit 5c9f423e (2021-08-24): fix #270