mooltipass/minible #271: Use unambiguous typeface

hijackeel, 2021-07-14:
### Missing feature
When viewing passwords on the display:
- Zero and capital oh look the same.
- Lowercase ell and capital aye look the same, and similar to the numeral one.

### Justification
It is very frustrating when entering a long randomly generated password manually on a device. 

### Workarounds
1. Open Multicute.
2. Download password.
3. Copy to text editor.
4. Change font.
5. Take photo on phone.
6. Carry phone over to device.

MaplesThroughWindows, 2022-04-19:
This shouldn't need a thumbs up.  Using a non-ambiguous typeface for something as important as this should be taken as standard.  This isn't an enhancement request, it's a bug report.

My1, 2023-02-17:
can't agreee more. imo the small L should have a small tail after it like several monospace fonts offer, equall the big i should have 2 bars top/bottom also a feature of many mono fonts. also in many fonts the big o is just wider than the zero, although if we just wanna make a mono space font the zero could be filled up with something, be it a dot, a slash or whatever really

CGuy-1, 2023-05-30:
"0O1Il" Even in this example font Il or is it lI look the same. The 5 characters need to be visually unique. I grew up with a slash through the zero 0, capital i (I), lower case L (l) and the number one (1) should also be distinct.

hjkiefer, 2023-10-27:
I fully share the sentiment by the other commentors.

Please consider commitmono (I am not affiliated with the authors!)
https://commitmono.com/