mooltipass/minible #272: UK Extended keyboard problems

awilkins, 2021-07-14:
### Expected behavior

MiniBLE types circumflex (U+005E) correctly in UK Extended layout, as <kbd>Shift + 6</kbd>

### Actual behavior

MiniBLE types circumflex modifier <kbd>AltGr + 6</kbd>

- UK Standard result is `¾`
- UK Extended result is that the circumflex modifier is set
   - This usually results in the next keystroke being "eaten" or modified

Should produce the desired result (a `^`) regardless of whether you have UK Standard or UK Extended selected as input method on target machine.

### Step by step guide to reproduce the problem

- Create a credential with a `^` in it
- Set MiniBLE keyboard layout to UK Extended
- Type credential

### Moolticute Version - If Involved

### Firmware Version

AUX MCU version:  0.62  
Main MCU version:  0.73  
Bundle : 2

### Operating System

Ubuntu 20.04  

---

Correct use example of this key combo

- `Ŵ` should translate to <kbd>AltGr + 6</kbd> <kbd>Shift + W</kbd>

---

- Created a test cred
- Set password to `^`
- Typed the password into a `xev` window with MiniBLE HID
   - This was set to UK Extended
   - I have reconfigured to UK Standard which fixes **my** problem but not this one

Log below.

What is required for this keymap is keycode 50, or 60 (Shift_L or Shift_R), not keycode 92 (ISO_Level3_Shift or AltGr).

```
MappingNotify event, serial 37, synthetic NO, window 0x0,
    request MappingKeyboard, first_keycode 8, count 248

KeyPress event, serial 37, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683209, (112,-11), root:(2032,45),
    state 0x10, keycode 108 (keysym 0xfe03, ISO_Level3_Shift), same_screen YES,
    XKeysymToKeycode returns keycode: 92
    XLookupString gives 0 bytes: 
    XmbLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyPress event, serial 38, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683224, (112,-11), root:(2032,45),
    state 0x90, keycode 15 (keysym 0xbe, threequarters), same_screen YES,
    XLookupString gives 2 bytes: (c2 be) "¾"
    XmbLookupString gives 2 bytes: (c2 be) "¾"
    XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683240, (112,-11), root:(2032,45),
    state 0x90, keycode 108 (keysym 0xfe03, ISO_Level3_Shift), same_screen YES,
    XKeysymToKeycode returns keycode: 92
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683240, (112,-11), root:(2032,45),
    state 0x10, keycode 15 (keysym 0x36, 6), same_screen YES,
    XLookupString gives 1 bytes: (36) "6"
    XFilterEvent returns: False

KeyPress event, serial 38, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683256, (112,-11), root:(2032,45),
    state 0x10, keycode 36 (keysym 0xff0d, Return), same_screen YES,
"   XLookupString gives 1 bytes: (0d) "
"   XmbLookupString gives 1 bytes: (0d) "
    XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0xac00001,
    root 0x1e9, subw 0x0, time 25683272, (112,-11), root:(2032,45),
    state 0x10, keycode 36 (keysym 0xff0d, Return), same_screen YES,
"   XLookupString gives 1 bytes: (0d) "
    XFilterEvent returns: False
```

limpkin, 2021-07-14:
Wow, this is one of the best bug reports I have ever seen.
So in theory the mini BLE should be aware of dead keys. In the case of the ^, it will type a space to get rid of the modifier (which leads to some issues on Linux systems were the compose key isn't set, see #263 where you already contributed). I'm guessing in your case it isn't enabled?
Looking at the extended layout in https://en.wikipedia.org/wiki/File:KB_United_Kingdom_Ext.png, it seems to me that both altgr and shift combinations are OK (not sure though, I'm typing on a different layout right now).

Our algorithm at https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr_parser.py#L184 is detecting dead keys by detecting that (for example) ^ + <space> = ^ and therefore whenever the mini BLE needs to type a dead key, the latter will add the extra space to type (https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr_parser.py#L638).
However, in the case of UK extended (I need to look at other layouts, but am fairly sure you don't have this use case for french for example) there is a way of typing a dead key without typing the extra space (the shift combination you mentioned) and it can actually be found in the CLDR: https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-GB-t-k0-windows-extended.xml#L67 . But that means that depending if you want to type the dead key or the dead key + other vowel (â) then the ^ you're going to type will need to be different (therefore "splitting" the logic at https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr_parser.py#L644 ).

I'm wondering if that's quite a tough one to implement without creating all sort of collateral damages on other layouts :/

awilkins, 2021-07-15:
On the description above it should work, but what I'm getting is 

- Input method on computer set to UK Extended
- First line MiniBLE UK Standard
- Second line MiniBLE UK Extended

```
test	^
test	
```

Second line you can watch the MiniBLE type the dead key and then "eat" it with the `0x0d` (see `xev` log above), so the space isn't happening.

Curiously <kbd> Shift+AltGr </kbd> <kbd> Shift+6 </kbd> <kbd> Space </kbd> works for either layout, but this seems to be the compose feature, even though my settings claim I have compose off ; the compose char appears in the editor.

[image]

Just to be sure it's not trimming ..

Actual cred `^ prefix`, MiniBLE set to UK Extended

```
test	¾ prefix  # UK Std. input method
test	^prefix    # UK Extended input method
```

Change cred to `^WIBBLE`

```
test	¾WIBBLE  # Std.
test	ŴIBBLE    # Ext.
```

limpkin, 2021-07-16:
I think I'm getting close to a conclusion:
- we need to handle the transform="no" property of https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-GB-t-k0-windows-extended.xml#L67 to always use the AltGr + 6 combo to keep our logic the way it is (sending a space after a detected dead key)
- for UK standard, ^ is NOT detected as a dead key (see https://github.com/mooltipass/minible/blob/master/scripts/keyboards/cldr-keyboards-37.0/keyboards/windows/en-GB-t-k0-windows.xml): does that mean it is impossible to type ê with a uk standard layout @awilkins ?

However that doesn't explain the second line of your first test.... but anyway we can just add that transform="no" check to our cldr parser and test that way :)

awilkins, 2021-07-18:
> does that mean it is impossible to type ê with a uk standard layout

Not in an ISO Standard way I think...

On Windows you can do the trick where you hold Alt and type out a number on the numpad (my new keyboard is a TKL, but hey, all the keys on the Mooltipass are pretend ones). e.g. e-circumflex would be <kbd>Alt+0234</kbd>

On Linux I think <kbd>Shift+AltGr e ^</kbd> will work, as would <kbd>Ctr+Shift+u 0 0 e a Enter</kbd>

limpkin, 2021-07-27:
@awilkins did you update to bundle 3 and verified that things are now working? :)

awilkins, 2021-07-28:
I've been busy!
Happy to try it, but the bundle issuer has given me a 0_to_3 bundle, and I'm already on bundle 2 from the knock detection issue - can't apply it, just getting an error.

limpkin, 2021-07-31:
@awilkins did I give you the right upload password? if no, send me a private message :)