mooltipass/minible #273: Implement WebHID for Mooltipass management via extension for Chrome and/or ChromeOS

spoelstraethan, 2021-07-17:
### Missing feature
Unable to manage the Mooltipass stored credentials on ChromeOS, since the HID interfaces are grabbed by ChromeOS and aren't available to pass through into Linux apps (Crostini).

If using WebHID an extension or website could request access to the specific Mooltipass USB descriptor and interact with the stored credentials rather than requiring manual login on the Mooltipass device itself on Chromebooks.

### Justification
Chromebooks are extremely useful and increasingly capable devices, and they are outselling almost every other type of computer by a large margin due to their extension use in education settings and for those on a tight budget. Being able to interact with the stored credentials rather than requiring manual login on the Mooltipass device itself on Chromebooks would be huge, and also keeping credentials stored separately from your Google account, because saving passwords in Chrome only encrypts them with your account password, and if someone gets your username and password (phishing), then they could get ALL of them, vs using the Mooltipass to keep them safely separated as well as being able to add it as a security key to your account.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

Manually selecting credentials on the device works and will type via USB or Bluetooth on the ChromeOS devices I've tested with.

limpkin, 2021-07-18:
@spoelstraethan "Unable to manage the Mooltipass stored credentials on ChromeOS, since the HID interfaces are grabbed by ChromeOS and aren't available to pass through into Linux apps (Crostini)." : could you tell me where I can read more about this? I do remember seeing very interesting things when sniffing USB communications.
Please correct me if I'm wrong, but my main issue with WebHID is that any website could just interact with the MiniBLE the moment the user would give its permission, right?

ethansTJ, 2021-07-18:
The permission is granted per site so it isn't automatic for all sites, and it will likely be something that browsers will require prompting EVERY time a site requests the permission similar to most other hardware interacting APIs like WebBLE and WebSerial and WebMIDI.

spoelstraethan, 2021-07-18:
The permission is granted per site so it isn't automatic for all sites, and it will likely be something that browsers will require prompting EVERY time a site requests the permission similar to most other hardware interacting APIs like WebBLE and WebSerial and WebMIDI.

spoelstraethan, 2021-08-02:
@limpkin One other route I was attempted but not having much luck with was modifying the Mooltipass Chrome App to add the USB Vendor ID and Product ID to see whether it could talk to the MiniBLE, but it is possible there are more checks in the source code for those that I missed beyond just in the `manifest.json`.

I know Chrome Apps are "going away", but it was mostly as a stopgap for managing the credentials from a Chromebook which is my primary device, and dragging out my Windows or macOS machines and ensuring they are charged is a hassle just to add a new credential that I end up using manually on the Chromebook anyways since the Mooltipass extension can't talk directly to the device to let it know what website URL I'm at and needs to go through Moolticute, which can't see the MiniBLE from the "jail" that Crostini/Linux apps on ChromeOS are run from.

limpkin, 2021-08-03:
@spoelstraethan I'm afraid the low level protocol for our chrome app has completely changed :/. Ideally, someone with enough time could make an app that provides the raw basics (credential recall and storage).

spoelstraethan, 2021-08-04:
https://github.com/robatwilliams/awesome-webhid#status

Looks like WebHID should be available in Chrome/ChromeOS without any extra flags since March 2021, so I might have to play with talking to the MiniBLE from an app/extension to see how rough it will be.

spoelstraethan, 2021-09-09:
Perhaps it would be possible to use QT for WebAssembly to help out with making updates to the extension to allow it to stand in for the full desktop version?

https://doc.qt.io/qt-5/wasm.html
https://doc.qt.io/qtcreator/creator-setup-webassembly.html