mooltipass/minible #274: TOTP time seems to not sync via USB if BLE is enabled (and not connected to any devices)

kbembedded, 2021-07-19:
### Actual behavior
When attempting to login with a TOTP auth, it kept giving me failed passcode errors. Using https://totp.danhersam.com/ that I already had set up as a TOTP test account, I saw that the codes the miniBLE was giving me were wildly incorrect; well outside of 90 seconds. moolticute was all happy with the miniBLE connected (though I didn't review logs).

I decided that I was going to reboot the miniBLE, but when I unplugged USB I got the "BLE disabled" message. I logged back in, disable BLE manually, connected USB, and observed the TOTP codes immediately matched https://totp.danhersam.com/ as expected.

At the time, the sole BLE pairable device I have was off so BLE was just on and flapping in the breeze without a moolticute client to connect to.

### Step by step guide to reproduce the problem
I need to try to repeat this a few times to see if its actually an issue. I just wanted to throw it up here for reference and will try to do some more detailed testing soon.

### Firmware Version
AUX MCU version:  0.58
Main MCU version:  0.68

### Operating System
- Windows 10

limpkin, 2021-07-19:
Hello there!

That's odd indeed... Please let me know how your experiments go. 
In the short term future we want to implement a new mechanism to improve time keeping (using the BLE transceiver 32k crystal)

kbembedded, 2021-07-20:
I spent some time trying to reproduce this and was not able to. No matter the BLE status as soon as it connected to moolticute via USB the TOTP time always synced. I probably just ran in to some strange hiccup. If I see it come up again I can attempt some further debugging. But keeping internal time I think would address this overall regardless.

limpkin, 2021-07-23:
yep :). I'll close this issue as the real issue may not have been linked to the USB / BLE interfaces...

closed 2021-07-23 by limpkin