mooltipass/minible #275: update minible from bundle verison 2 to 3 using 0 to 3 firmware

GeekAnnoyed, 2021-07-21:
Ive just got an email from you and downloaded the firmware
but I noticed it's expecting version 0 not my ble's current version of 2
I got the 2 update after emailing you from the main mooltipass website

is it safe to apply this update?

Many thanks

[image]

limpkin, 2021-07-21:
it is! but you'll need another password :). 
Can you send me a private email?
BTW: please update your moolticute version :)

closed 2021-07-21 by limpkin

schlomie, 2021-09-22:
Hi @limpkin 

About two weeks ago, I sent an email to support@themooltipass.com asking about this very issue - but going from 0 to 4.  The password on my bundle upgrade file does not work.

Please advise.  Thank you.

limpkin, 2021-09-22:
@schlomie could you send us another message again? it seems we missed your message :/

schlomie, 2021-09-22:
Sent - thanks!