mooltipass/minible #277: upgrade ble to v3 and no longer can connect to browser

GeekAnnoyed, 2021-07-22:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
visit e.g. github and expect my mooltipass to request to unlock/send username/password
### Actual behavior
visit github.com 
nothing happens

### Step by step guide to reproduce the problem
install Moolticute
install Firefox extension
visit website I know I have a password saved in my ble

### Moolticute Version - v 0.50.0

### Firmware Version
AUX MCU version:  0.64
Main MCU version:  0.75
Bundle Version 3

### Operating System
Mention if you are using either:
- Windows 10
- and 
- Ubuntu kubuntu

 just as a side note It stopped working when i upgraded to bundle 3 yesterday
i can still "type" by selecting usb/bluetooth and using the virtual keyboard 

not sure how to get any logs as I'm not sure what's at fault here.

  
Other operating systems are not supported.

### Mooltipass Extension
- If you're creating an issue to report a website incompatibility, please use the "Report incompatibilities with this website" item in the extension menu.
- If you're creating an issue to report a problem with our extension, please create another issue here: 
https://github.com/mooltipass/extension/issues

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-07-23:
that's extremely odd... could you try restarting your computer?
I have a feeling that should fix things...

GeekAnnoyed, 2021-07-23:
I've actually installed ubuntu after the issue started, as I thought it was my AppImage/ Manjaro install that was causing the issue

GeekAnnoyed, 2021-07-26:
Okay, the craziness carried on, Things like when I scrolled up on the password list, it would reset the device. So I went extreme and removed the user and stared afresh. And all issues seem to be sorted now.

closed 2021-07-26 by GeekAnnoyed