mooltipass/minible #278: Forced into left-hand mode when no power attached

alanrick, 2021-07-27:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Display flips when the mooltipass BLI is upside down.

### Current behavior
The display only flips when attached to a power supply.
If attached to a power supply, the mooltipass prompts for left-hand or right-hand mode when it is flipped upside down. However, when relying on the BLE internal power no such prompt is displayed, the display is not flipped and I am forced to either use left-hand-mode or to use it with the display showing the characters upside down.

### Step by step guide to reproduce the problem
Insert card into mooltipass mini while it is not attached to any usb cable. You will be prompted for the PIN, but don't enter anything. Instead,  rotate the moolticute by 180 degrees so that the text is upside down. No query appears about using in right-hand mode.

### Moolticute Version - If Involved

### Firmware Version
AUX MCU version:  0.62
Main MCU version:  0.73
Bundle version: 2

### Operating System
None - no cables attached and blue tooth is off.
  
Other operating systems are not supported.

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-07-27:
Hello,
Hmm that's odd... could you try without inserting the card?

alanrick, 2021-07-28:
I think I've worked out the behavior. It's not a bug, but I found it confusing and jumped to the wrong conclusions. It's not related to the power supply.

The algorithm seems to be:
a) On power up, if card inserted, prompt pin, and then afterwards prompt for l/r-hand mode if the display is upside down.
b) On power up, if card not inserted then prompt for  l/r-hand mode if the display is upside down.
c) If no l/r-hand prompt was displayed after the last power up, then prompt l/r-hand mode whenever the display appears upside down. _However_, 
d) once the l/r-hand prompt has been rejected (answered with an X) then don't prompt again until the next power-up.

Now that I see the pattern I'm no longer confused and can live with this. Apologies for the false bug report.

limpkin, 2021-07-31:
actually there's a timer to not spam the user... it is set to one minute... could you check?

alanrick, 2021-08-01:
It's a bit more complicated than simply 1 minute. If I switch it on after powering off and answer "yes" to changing to switching the mode then it will prompt again when I flip it by 180 degrees even after 2 minute.

It would be sufficient if the instructions stated that. 
"For the first minute after powering on, you have the option of toggling left or right-handed mode by turning the display so that the text appears upside down".

That would help people like me who searched in vain in the moolticute app (with and without the card inserted).

limpkin, 2021-08-01:
Let's do it like this then :)

closed 2021-08-01 by limpkin