mooltipass/minible #298: [Website] Mooltipass firmware updates

KDederichs, 2021-09-12:
I think the firmware update website is slightly bugged?
When I try to download an update it'll give me the 0 to 4 bundle, even though I'm on 3 already. (Clicked the download link in the new moolticute so that was passed in as well).

I know you can just give out another password via email to enable those bundles regardless but it would be less work for us and you if the website spits out the correct file in the first place :)

cvantum, 2021-09-13:
Same here. 

I am on Bundle 3 and not able to upgrade to 4 since this firmware required a version 0 to start from.

KR

awilkins, 2021-09-15:
Moolticute now supplies your current bundle version to the web page in a query param, but the web app doesn't put the correct hash in the filename.

The 0_to_N bundles are good for all updates but the hash differs depending on the bundle you're starting on, so the "upload password" fails (I presume there's e.g. a unique salt in each bundle version and without knowing it you can't generate the "correct" hash).

limpkin, 2021-09-15:
Until we update the page to take into account the bundle version passed to it, I just updated the files to include the other passwords :)

cvantum, 2021-09-15:
I can confirm that the current files is working fine. I was a bit confused with 0_4_XXXXX_3_4_XXXXX naming but the mooltipass app were able to update the firmware.

Maybe you should implement a third input form at your update-page to ask for the current installes build.

hilakata, 2021-09-15:
No dice in my case, unfortunately.
Checked multiple times that I used the correct (second) pwd in the filename. 

Still get 'Upload bundle finished with error'.

limpkin, 2021-09-15:
@hilakata can you double check you inserted the card the wrong way around?

hilakata, 2021-09-15:
Shoot, you were right. Officially owner of the noob title for the day...
THX!

limpkin, 2021-09-15:
no worries, you're far from the being the first one :)

ExpectationMax, 2021-10-17:
Having the same issue here 😕
The upload fails presumably due to an incorrect hash.

limpkin, 2021-10-17:
@ExpectationMax did you select the hash that matches the bundle version currently on your device?

ExpectationMax, 2021-10-17:
Good point! Didn't realise there are two hashes in the bundle name 🤦 

Now everything worked for me. Thanks!

limpkin, 2021-10-17:
no worries... in any case, we need to update this page to take into account the bundle version supplied to it :)

johncollaros, 2021-10-20:
Hi, I have a minible and its constantly throwing errors (002 and 005) after I enable bluetooth and nearly becomes unusable.
I have tried upgrading the firmware using moolticute on Linux and Mac and I get the 'Upload bundle finished with error'.

The bundle file downloaded has both 0 to 4 and 3 to 4 in the filename.

Based on the manual, if it doesnt show my bundle number in moolticute, I am to assume it is bundle 0, so i am using the bundle 0 to 4 key. I have also tried the other key and I still get the error message.

What am I doing wrong?

Any help would be appreciated.

Thanks,

John

sonofapharmacist, 2021-11-03:
Can you confirm that the way to turn around the card the wrong way is to flip it over so the chip faces up and is still inside the reader?

For my current firmware build (2) should I use the password that is after _0_to_4_ ?

If all this is correct, then I am still not able to update firmware. 

An image of my current firmware, etc.
[https://imgur.com/a/HQ1ofLO](url)

limpkin, 2021-12-09:
@sonofapharmacist I confirm indeed (sorry for the delayed answer).
The new website now takes into account the current bundle version :)

closed 2021-12-09 by limpkin