mooltipass/minible #299: Feature request: device-specific password to erase its memory contents

limpkin, 2021-09-15:
... using an automatically incremented counter so that the same password can't be used twice

MaplesThroughWindows, 2022-03-22:
Are you referring to a "duress password" - i.e. when forced to supply or enter your device PIN, it performs some other action that allows you to maintain your security, at least to a degree?

I thought [these comments](https://news.ycombinator.com/item?id=30584743) were interesting:

> 
> franga2000:
> Duress passwords aren't there to stop brute-forcing, they're there to save your life in the event of wrench-based cryptoanalysis. Think "log into this system or I shoot you". Using a duress vs regular password should look identical and give away as much information/access as you can afford so your odds of being found out (and subsequent shot) are as low as possible.
> 
> Even if configured as sef-destruct (in cases where you value system security more than your life), it's still better than a basic panic button (ideally you want both). You can point a gun at someone and tell them to keep their hands away from a button, but eventually you'll need to have them open a locked door for you and in that moment they are free to type anything on the keypad, be it "open door and call the cops" or "blow up the whole building" and you won't know what they typed until it's too late to stop them.
> 
> djur:
> I'm aware of that, which is why the comment you're replying to suggested that duress passwords are not a good defense against brute forcing.
> 
> giantg2:
> True. I think that's mostly an edge case though. The vast majority of the time it's the police forcing a person to unlock a device.

TL;DR: It may not be wise to simply erase a Mooltipass's contents, because you risk angering those forcing you to reveal them - in some (most/all?) jurisdictions, for example, that may consitute a crime if the ones forcing you are law officers.  

A PIN that displays a bogus or curated sub-section of the Mooltipass's contents might be preferable.  I suppose there would be no harm in it _also_ deleting the non-bogus/non-curated contents.

Or maybe you just meant a password for performing an ordinary reset, in which case you can ignore this comment.  :)

limpkin, 2022-03-22:
I'm not referring to that, I think duress PIN is captured in another issue.
This issue is to provide a feature so that when someone wants to sell his device to someone else, that the initial user could erase the Mini BLE storage contents (eg: erase ALL users databases on the device)

d3t4ash, 2022-04-18:
> I'm not referring to that, I think duress PIN is captured in another issue. This issue is to provide a feature so that when someone wants to sell his device to someone else, that the initial user could erase the Mini BLE storage contents (eg: erase ALL users databases on the device)

I wont buy or sell this kind of items in the second hand market. The software might be tampered and make the device feel insecure.

limpkin, 2022-04-18:
this is everyone's choice :). If someone tampers with the mini ble's firmware, then our security challenge feature will report an issue.

opie4624, 2022-04-29:
Or maybe you moved house and you haven't found your smartcards and you want to setup the device again with the blank cards you _have_ found.

Definitely not my specific case... 🙄

My1, 2023-02-17:
would one even need a device specific password for wiping? I mean ideally one should have a backup of their dbs anyway, so a wipe option within MP would be nice for that

CGuy-1, 2023-10-13:
I like the idea of wiping the good db and presenting a bogus one in its place. The user could setup the bogus db with entries that look real for the user. Not all police or other agencies will go get a warrant first, so killing the main db and presenting the backup bogus one is a great idea.

As for wiping all DBs on the device, something similar to the challenge key for testing the S/W could be setup. Mooltipass Co. would also need to allow a user unassign the device so that the new user could register it with their ID/e-mail address.