mooltipass/minible #300: Feature request: extract TOTP seed

limpkin, 2021-09-16:
Not sure if using a specific USB message or displaying the seed on the display....

Yomark1, 2022-01-06:
Perhaps it's also possible to add a menu entry in the Browser extension "Fill TOTP" al well.

neeksor, 2022-10-04:
Any updates on this? I'd like to export some TOTP secrets to migrate to another device.. The process to reset TOTP on account providers requires acts above my pay grade. :D

limpkin, 2022-10-04:
I'll see if I can fit it in bundle v11 :)

limpkin, 2023-02-17:
I'm afraid I didn't got to it... should be fairly simple to implement for a new contributor though :)

CGuy-1, 2023-05-29:
I have been keeping the seeds in the "Notes" section but this isn't always possible if I don't have my laptop with Multicute running. 
Nice to have.

limpkin, 2025-04-22:
dang it... I'm afraid not at this moment... though it should be easily coded