mooltipass/minible #304: Support hmac-secret FIDO2 extension

barathrm, 2021-10-10:
### Missing feature

https://fidoalliance.org/specs/fido-v2.0-rd-20180702/fido-client-to-authenticator-protocol-v2.0-rd-20180702.html#sctn-hmac-secret-extension

*NOTE* systemd-cryptenroll doesn't seem to (?) detect the minible as a valid fido2 device, so you may have to specify it manually like so: 
```
sudo systemd-cryptenroll --fido2-device=/dev/hidraw10 /dev/disk/by-id/<partition id>
Specified device /dev/hidraw10 is a FIDO2 device, but does not support the required HMAC-SECRET extension.
```

### Justification

My specific use-case is that this is now one (apparently) very easy way to use the minible to decrypt LUKS-encrypted volumes. Here's a guide for it: 

http://0pointer.net/blog/unlocking-luks2-volumes-with-tpm2-fido2-pkcs11-security-hardware-on-systemd-248.html

Relevant man pages for systemd-cryptenroll and crypttab

https://www.freedesktop.org/software/systemd/man/systemd-cryptenroll.html
https://www.freedesktop.org/software/systemd/man/crypttab.html#

Yubikeys seem to support this. 

### Workarounds

I have to/can manually find and enter the credential using the minible.

My1, 2023-02-17:
hmac-secret is a pretty nice thing and iirc also needed for AAD+FIDO stuff which is pretty nice.

My1, 2023-08-04:
side note: this issue is also blocking #353 (mislabeled currently as it's rather about credprotect which SSH asks for in relation to resident keys)

credprotect needs CTAP2.1 and CTAP2.1 requires hmac-secret