mooltipass/minible #306: Moving from Mooltipass to Mini BLE, characters are not same after importing

wisemonkey, 2021-10-31:
### Expected behavior
Create Backup on local machine with old Mooltipass and old smart card.
Turn on mini BLE, insert old smart card and import database
Clone old smart card, insert new smart card and finish cloning
Use mini BLE with new smart card without issue

### Actual behavior
When logging in with mini BLE characters entered are not correct.
(Maybe keyboard is not correct)
mini BLE enters 'Q' instead of 'A'

### Step by step guide to reproduce the problem

### Moolticute Version - If Involved

### Firmware Version
AUX MCU version:  
Main MCU version:  
Not sure about FW versions

### Operating System
Mention if you are using either:
- MacOS

wisemonkey, 2021-10-31:
I was bit impatient since I didn't find how to figure out keyboard.
Advanced mode -> change keyboard to proper keyboard seems to have fixed the issue.
I'll re-open the issue if I see discrepancy

closed 2021-10-31 by wisemonkey

limpkin, 2021-10-31:
Hello!
Please note you can also change the device's keyboard layout in moolticute settings :).
We highly recommend reading our user manual.. we spent quite some time on it :)