mooltipass/minible #307: Importing CSV add an "approve all" option

hon1nbo, 2021-10-31:
### Missing feature
When importing a CSV to a freshly configured device, it prompts for adding the credential on every single entry. For those of us with a large presence of unique credentials across forums, this is effectively pressing a button, waiting two seconds due to the unskippable animation, and repeating hundreds of times (in my care, over 700 for what is not even my complete list)

### Justification
Tell us why you would like to see this feature added.
For users with existing password databases this will make importing far less painful. Given that these entries did not exist previously, there should be a way to accept them without needing to approve a "modification"

This could possibly be set to only work on a device that does not currently have entries, as this would mitigate the potential for managing entries that could be conflicts. However, prompting when there is a conflicting entry seems appropriate.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?
a sore thumb

limpkin, 2021-10-31:
closing this issue as the user manual states that the device should be set in simple mode when attempting this (section H) Advanced tab)

hon1nbo, 2021-10-31:
Received an email on separate channel, the documentation is just missing.
Device in simple mode does not prompt.

Can we update the documentation for CSV format requirement and caveats?

hon1nbo, 2021-10-31:
The manual says nothing about that affecting CSV importing. This really should be more clear

hon1nbo, 2021-10-31:
For the manual section on device setup, the wording is:

"If you’re a novice user and prefer your device to be as automatic as possible, by, for example, automatically approving credential requests for you, approve the “Simple Mode” prompt.  You can always change your choice later. "

That "automatically approving credential requests" reads for native English speakers as credential requests sent to the device, not data entry.

The section for advanced mode, section h, has no mention of Bulk Imports being affected.

Section h.e specifically reads:
"This menu item allows you to specify whether you’d like to be prompted to save credentials entered during management mode. "

This again, is not explicit that a bulk action is not approved as a bulk action. The inherent nature of a bulk action means that anything other what's expected needs explicit context.
What user would reasonably think that a bulk import operation would require every entry to be manually validated?

The expected behavior based on the text of the manual would be:
simple mode: just accept whatever without prompting
advanced mode: prompt to accept a credential (individual), or a _set of credentials_

hon1nbo, 2021-10-31:
Ah, you mean under the application header H and not the Device header H.  The application header H, covering the moolticute application advanced tab, still does not indicate implications as to what advanced vs simple does. For users that want to ensure prompting on actions this still could use clarity.

limpkin, 2021-10-31:
fair point... we'll therefore update our user manual :)

hon1nbo, 2021-10-31:
Regarding my device,

all scrollwheel stopped post pin-entry after the update. MC does not show
card data anymore either, but the battery now says depleted.

still getting Contact Support 002 sometimes but not consistently.

commit cd483948 (2022-03-17): user manual update for android app, fix #307, fix #325
closed 2022-03-17 by limpkin