mooltipass/minible #308: MiniBLE hangs then 002 contact support, then unresponsive

Mixer9, 2021-11-07:
Use one of the two templates below and delete the other. All issues not following this template will be directly closed.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Expect MiniBLE to work as designed, expected to use menus and select credential to input to MAC login screen

### Actual behavior
After choosing login, then moving to a credential MiniBLE hangs and then gives 002 contact support error.  After this it reboots then the wheel is unresponsive, you have to shutdown the MiniBLE and reboot , then it works as expected.

### Step by step guide to reproduce the problem
Let your MAC lock with screensaver, then attempt to login after some hours of being locked.  This is a rare occurance, about once a month.

### Moolticute Version
0.52.0

### Operating System
Mention if you are using either:
- MacOS  Big Sur 
  
Other operating systems are not supported.

### Mooltipass Device
Mention if you are using either:
- The Mooltipass Mini BLE

[Debug.txt](https://github.com/mooltipass/moolticute/files/7460803/Debug.txt)

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2021-12-09:
@Mixer9 can you updated to bundle v6 using https://www.themooltipass.com/beta_updates ?

sorry for the long delay... i didn't get github notifications :/

Mixer9, 2021-12-10:
yes I updated to bundle v6, we will have to wait and see if it happens
again.
Chris

Retroman445, 2021-12-16:
Yes it happens with Bundle v6:

[image]

But it seems to be related to Bluetooth. I get #002 and #003 Contact Support when enabling bluetooth and trying to pair or when paired, just using it.

I want to supply some more debugging info, but how can I supply that from the Mooltipass?

limpkin, 2021-12-16:
And here I was happy with no reports of 002 on bundle v6 for 2 months.... :P
Does bluetooth communication work well on this device? I want to make sure this isn't device related.

Retroman445, 2021-12-16:
> 

Never did particualry well with bluetooth, lots of disconnects from the start, waited for a new firmware, maybe it got resloved. But alas. 

Had battery problems also, but have resolved them with repeating recondition sessions.

limpkin, 2021-12-16:
Do you mean the device keeps disconnecting to your computer for no reason, while it is close to it?

Retroman445, 2021-12-16:
> Do you mean the device keeps disconnecting to your computer for no reason, while it is close to it?

Yes, very rapidly.

limpkin, 2021-12-16:
I'm guessing this behavior is only for this device and not your other one?

Retroman445, 2021-12-16:
I wish I had two. ;-) I only have this one.

limpkin, 2021-12-16:
Time to send you another one then... send me your address in private, with a quick video of the connection/disconnection issue?

Retroman445, 2021-12-16:
Ok, I will anwser in the e-mail to you with the address. If I can get it to pair again I will post the video here on github.

Retroman445, 2021-12-16:
https://user-images.githubusercontent.com/51086501/146358284-c1e7a24c-c56b-4707-9646-453b0c4ca29d.mp4

Retroman445, 2021-12-16:
https://user-images.githubusercontent.com/51086501/146358744-c3569256-6107-445c-895b-8a94194755e5.mp4

Less rapid, but it fails at the end. The Mooltipass is located next to the laptop. Other devices connect without issue.

limpkin, 2021-12-16:
yeah that really looks like a hardware-centric issue. I'll send you a new device.

BGameiro2000, 2022-01-02:
Hi, I have the MiniBLE for about a week and it has worked great.
Today I used Bluetooth for the first time and didn't disconnect it before it locked, a coupled of hours later and I got the "#002 contact support" message. It changed rapidly between showing the message, not showing anything or showing the menu unresponsive. 
A forced shutdown and disabling Bluetooth seem to have fix it for now.

I'll update in case it becomes something regular and not a one time thing.

Versions:
- Moolticute: v0.53.7
- OS: Arch Linux 5.15.11-arch2-1
- Aux MCU: 0.66
- Main MCU: 0.77
- Bundle version: 5

limpkin, 2022-01-02:
Hello Bernardo,

Could you update to bundle v7 using themooltipass.com/beta_updates/ ?
Bundle v6 fixes the #002 issue :)

Mathieu

BGameiro2000, 2022-01-02:
I get "Invalid serial number / email combination".

I bought it from Tindie with the email inbox@bgameiro.me

limpkin, 2022-01-15:
closing this issue as fixed by bundle v6 :)

closed 2022-01-15 by limpkin